AlphamaxMedia/netv2-fpga #3: Alternate option to Magic Mirror

JpEncausse, 2019-02-04:
Hello,
Is there a workaround to use something else than Magic Mirror ? How underlayer works ?

- Declaring a simple HTTP endpoint (local or distant) with a transparent pixel like NeTV1
- Running another NodeJS server like Node-RED with Dashboard module very popular

Thanks!

bunnie, 2019-02-04:
Yes, you can run anything you like actually. You just need to kill Magic
Mirror, and then start the UI engine of your choice.

Magic Mirror is convenient because it renders everything you don't want
to see as black, so the chroma is set to black and the letters just
composite nicely over the image.

However, if you fire up a chromeless browser, you can just run HTTP and
it works the same -- any area that's black is transparent.

In other words, the compositing engine doesn't care what middleware you
use to render the UI. MagicMirror is just the simplest turnkey solution
for people who don't want to write e.g. HTML and just plug in an
existing set of widgets.

thanks,

-b.

bunnie, 2019-03-21:
I think this answers the question, so closing...

closed 2019-03-21 by bunnie