AlphamaxMedia/netv2-fpga #4: Change key colour

xer0design, 2019-03-14:
Hey,

Just received my netv2 a few days ago, loving everything so far, the key set to black is causing a few issues though, mostly edges showing as green or pink. Can you advise on how to change the key to FF00FF or something rather than 000000 to see if that might help, it'll also let me design overlays with black in them.  Thanks!

bunnie, 2019-03-14:
Right...I had a feeling I'd get a request like this. I've prepared a "testing" version of the FPGA image to support this. To try it out, please check out the branch chroma-set and go to the testing-images directory and run the testing-fpga.sh script there to apply the latest testing firmware. Then go to flterm and open an interactive shell and set the chroma values to your desired levels. 

(NOTE: if you have the 100T version, you'll have to wait about 8 hours after you receive this to do the following, I've kicked off the automated build to create all the image variants but that takes about an hour and I'm heading to bed now, I'll check in the other variants in the morning when I wake up. So this will currently only work if you have the 35T version)

More specifically, for a chroma of 0xff00ff:

```
cd ~/code/netv2-fpga
git pull
git checkout chroma-set
cd testing-images
./testing-fpga.sh

pm2 stop netv2-status
cd ~/code/flterm
./flterm --port /dev/ttyS0 --speed 115200
# now inside flterm
json off
chromalo 0xeb00eb
chromahi 0xff10ff
chromapol 1
json on
# hit control-c to break out of flterm
pm2 start netv2-status
```

The pm2 commands keep the netv2-status daemon from gobbling up serial console characters, which makes debugging on the terminal hard. The json off/on command turns off the json telemetry reporting via the interactive shell which also makes it hard to debug.

Notes on the new chroma commands:
```
chromapol, when 1, means pixels between upper/lower map to video
chromapol, when 0, means pixels between upper/lower map to overlay
chromalo sets the lower bound for chroma testing as hex 0xRRGGBB
chromahi sets the upper bound for chroma testing as hex 0xRRGGBB
```

both values are inclusive of the setting, so using the same value passes just that one color.

Note that the Raspberry pi does some sort of funky gamma correction by default -- so the colors you specify to draw aren't exactly the colors sent to the screen. I'm not sure how to turn it off, but that's why the chroma values don't exactly line up with the values you intend them to be. I had to narrow into the chromalo/chromahi values by trial and error more or less, and I worry that the color clipping may even be monitor-dependent, in case they are doing something like reading back the recommended gamma curve of the monitor and then compensating for it before sending the values on.

Note that the default settings would be chromalo 0x141414, chromahi 0xffffff, chromapol 0 -- in other words, most of the video UI colors are rendered, but everything that's almost black gets turned into video. This is a bit easier to work with in terms of the post-rendering gamma correction.

bunnie, 2019-03-14:
OK, the 100T-pcb version is checked in, so this should work if you have any of the quickstart kits. The only one pending build is the 100T-cable.

xer0design, 2019-03-14:
Thanks bunnie, that worked perfectly for changing the key. I'm having issues with black showing as green though; white is tinted slightly pink also, is that an issue on my end or something you've seen before? See attached images of what vnc sees and what the netv is outputting. I'm on the 35T kit. 

https://imgur.com/a/qvV0Jyy

bunnie, 2019-03-15:
ah yes. This is because your video source has chosen YCrCb colorspace, and the monitor is expecting that, but the NeTV2 is compositing RGB.

Can you force the colorspace of your video source to RGB?

xer0design, 2019-03-15:
For testing at the moment I'm just using my laptop as the source, i can change the colour space so I'll try that. However, my main use for this is a broadcast TV box but due to that only outputting 1080i/720p I've been unable to test using that. I should probably open a separate issue I assume but how can I switch netv to one of those two signals? I tried setting the pi to match the input (ie 720p) hoping maybe having them match would fix it but sadly it does not, is there a command I can run somewhere to switch the source settings over?

bunnie, 2019-03-16:
I've got a new branch at

https://github.com/AlphamaxMedia/netv2-fpga/tree/ycbcr

This gives you the option to switch to ycbcr color space. It's very beta at the moment, I suspect there could be some tweaking that has to be done on the alignment of the color space conversion path and the chroma key masking computation, it's tricky to get it exactly lined up.

To use this basically follow the same procedure as above, but do "git checkout ycbcr" instead of "git checkout chroma-set".

All the chroma set commands are available, but there is also now a new command called "chromamode". Passing it an argument of 1 will set to YCrCb, 0 to RGB.

Would love it if you could confirm if this works for you or not. 

As for the 1080i/720p modes, please open a separate issue. The hardware is capable of supporting it but the firmware support for this is much much trickier. In implementing NeTV2 I had to basically rewrite entire portions of the litevideo interface from LiteX because the original implementation did not pick optimal MMCM/PLL settings and the clock routing was a bit non-optimal, plus I had to completely redo the pixel delay sampling engine. At the moment these all require hand-tuned parameters specific to a given resolution mode, and for the campaign I specifically only promised 1080p support because I knew this would be a pain.

But please open an issue for the 1080i/720p modes, and I'll try to contemplate how best to refactor the code to try to accommodate it. I will also want to ask you some questions in that thread about the UI and so forth, in case the resolution detection engines fail to work what's the right fallback and so forth.

Also it'd be good to merge the chroma changes back into the mainline before trying to tackle the resolution modes, so let's try to close this issue first with a thorough testing of the chroma setting and color space conversion.

bunnie, 2019-03-16:
Oh, btw, what command are you using to switch the pi to 720p? for the life of me I can't seem to find a command line that can reliably force the Pi to a video mode without requiring a reboot.

xer0design, 2019-03-19:
Hey! 

Sorry for the slow replies on my part, I only have time for this in my off time so fit it in where I can. On the resolution command, I have the same issue, I just edit confix.txt and reboot or use raspi-config if I'm trying ones I can't remember the codes for. I'll open that resolution issue now also.

Anyway, as for the ycbcr branch, it appears to be absolutely flawless so far, thank you; completely fixed the  I'll start playing with all my test graphics and scenarios now and see if I can break it or spot any issues but this definitely looks great so far. I notice a slight issue with underscan of the NeTV2 signal, so I'll probably need to ask how to modify the bounds at some point but that's not a major issue at all, probably about 1-2% of the corners get pulled in.

bunnie, 2019-03-21:
The underscan is a little bit of a tricky problem to solve, it's due in part because the processing delay of NeTV2 makes it miss a few pixels on the left. Also if you open the scan window too wide you end up dropping into the sync regions. 

Anyways, you can use the command

``debug setrect <left> <right> <top> <bottom>``

to play around with the underscan.

Defaults are here:

https://github.com/AlphamaxMedia/netv2-fpga/blob/master/firmware/ci.c#L692

So it's 32 1888 10 1070 by default.

bunnie, 2019-09-11:
So...here's an update on the color space switching issue. Tonight I was trying to automate the detection of colorspace and switchover to YCrCb. However, it turns out the  info packets that specify the color space are also encrypted with HDCP, so...I don't think I can lawfully automatically detect and switch over to the correct color space due to section 1201 of the DMCA.

I'm going to have to put this feature development on hold until we have a further ruling from the US government on the legality of this activity.

xer0design, 2019-09-11:
Thanks for the update! Hopefully something can eventually be done about that stupid law.

If you're operating from Asia do you still have to abide by us law? I guess you do when you're shipping there right

bunnie, 2019-09-12:
I am an American citizen, even when I operate from Asia.

bunnie, 2019-09-22:
Key color changing is baked into the latest release. Closing.

closed 2019-09-22 by bunnie