AlphamaxMedia/netv2-fpga #5: Change input resolution from 1080p/60

xer0design, 2019-03-19:
Hey,

Basically, the main input source I plan to use for the NeTV2 is 1080i/50 rather than 1080p/60, 50 because I'm in Europe and interlaced because TV STBs can only output in 1080i or 720p, generally. While on the topic, 720p would also be a great option to have depending on other peoples uses and I'm sure someone somewhere would be using 480p or something else for some obscure use. The original NeTV was great as it detected whatever you had plugged in and matched it; I know the pi complicates things as that would probably also have the match the resolution of the NeTV for them to composite correctly.

I understand the complications of doing all this automatically are probably pretty large, a manual toggle would more than suffice as how often do you really change your input settings. It might also be a great help if it could be set to output something while the pi is still booting or if there's an input error, just something like "source mismatch" or "loading netv2" so you can tell what's going on until the pi boots - although when the code is loaded from the pi would I be correct in assuming this would mean I'd have to add an SD card directly into the netv for it to load files while it waits?

bunnie, 2019-03-21:
One question about changing input resolution.

I suspect the rapsberry pi won't allow resolution changes without a reboot. 

How do you feel about when doing 720p modes, keeping the raspberry pi in 1080p but having only the top left 1280x720 pixels used for the overlay (and the rest is ignored). This way the raspberry pi stays locked in 1080p mode, and to change the video overlay mode we're just adjusting the DMA parameters of the overlay readout.

xer0design, 2019-03-21:
That'd certainly be a viable solution. It could complicate things a small bit as generally what I'm doing is using my interface to call `cvlc -f file.xyz` or `chromium-browser --app=xyz.com --kiosk` (with all content set to scale full screen) so I'd have to figure out how to limit the viewport there. It'd be easy to change that in chromium (just frame the content in the corner I suppose) but I'd imagine vlc might be a bit more complicated to do. Maybe the centre of the screen might be better suited rather than top left? I'm not sure. 

I can make anything work I'm sure but thinking for product completeness this might be where you'd have the NeTV overlay "resolution mismatch please set your pi to 720p" over the top left section that displays for x seconds when it boots so that anyone not familiar with the setup will understand what's happening. Even a "please refer to wiki for information on crop" or similar.

I should be able to spend a few hours tonight playing around.

xer0design, 2019-03-21:
Actually, if you wanted to do the top corner thing, you could set the overscan right/bottom in config.txt to make the display 720p? That's a bit of a hack but it might be possible to change those settings on the fly? You're still outputting 1080p but effectively it's a 720p frame you're using

config.txt:
```
overscan_right=640
overscan_bottom=360
```

xer0design, 2019-03-22:
https://github.com/ukscone/set_overscan

xer0design, 2019-04-08:
Hey Bunnie, just following up to see it there's been any progress on supporting any other resolutions in any way at all?

bunnie, 2019-04-09:
Sorry, I haven't been able to touch this one yet -- tons of travel and sick for a couple days. I'll try to open some time to take another look at it but I need to figure out a way to build the PLL maps more efficiently than the first time I did it by hand. Turns out the PLL mappings provided by default in LiteX were off-target and didn't retrain the loop filters, and the example code from Xilinx on how to do it doesn't match up with what their proprietary design tool spits out, so I've resorted to reverse engineering the output of the design tool to create the PLL maps. It takes a bit of effort, is my point, so I need to clear out a good chunk of time to look at this one.

xer0design, 2019-04-09:
No problem thanks for the update!

xer0design, 2019-04-10:
Actually, just looking back, I understand based on your explanation that other resolutions are difficult and time-consuming to map and implement, but is interlaced vs progressive a similar challenge/would that be easier to enable? 1080i would also suit me perfectly for now for my use case so I can continue developing what I need out of it - also at 50Hz, of course, I'm not sure if that breaks things too? No pressure to get it done fast, of course, I understand this is in your own time, just eager to keep building!

bunnie, 2019-04-18:
OK, I've got ... a kind of 720p support built in. Can you give it a try and let me know if it works for you in any way?

https://github.com/AlphamaxMedia/netv2-fpga/wiki/Multiresolution-support

Turns out 1080i is more complicated because the modification to the DMA engine needs to get the interlace correct. So it's easier to start with just checking that stuff like frame sync is working before trying to throw in interlacing.

bunnie, 2019-04-18:
I thought about it a bit more and found a way to implement 1080i, I think. I don't have a lot of good 1080i sources to test on, though, so if you could give it a try and let me know, that would be great. 

Same link as above for instructions on how to obtain the 1080i version, I'm just updating the existing repository. Hopefully I didn't break 720p at the same time...

xer0design, 2019-04-18:
I'm away from the office today sadly so I can't test immediately but I'm looking forward, thanks for your work. For 1080i test sources, to my knowledge every single TV set-top box (ie cable or satellite TV) uses 1080i to conserve bandwidth, my source is Sky Broadcasting here in Ireland but for America, the equivalent would be Comcast, Dish etc. 

Alternatively, if you have another pi laying around, you can set config.txt to the below to set the pi to 1080i 60hz
```
hdmi_group=1
hdmi_mode=5
```
Or use `hdmi_mode=20` for 50hz

xer0design, 2019-04-18:
Sorry for closing the issue, my mistake.

bunnie, 2019-04-18:
I've got a DVD player configured for 1080i and it's working, but ... there's a lot of trickiness to the interlace. So when you get a chance lmk how it works on your hardware. Just FYI I'm probably going to context switch out of closing NeTV2 bugs for a bit, so no hurry on your side, just would be good to get some feedback queued up on how it works at 1080i and 720p.

Note that you do have to issue a command from the terminal to switch the modes, as noted on the wiki page!

xer0design, 2019-04-18:
I'll test it out as soon as I possibly can and let you know straight away; very eager myself to try it out so it's top of my list. Setting via command line is no issue whatsoever. Actually, if I use 720p, how does the pi have to be setup? Do I switch that over to 720 or are you using your earlier plan of using the top left area of the full output or how does it work?

bunnie, 2019-04-19:
The 720 is done using the top left area of the rpi still in 1080p mode, so you don't need to reboot the rpi.

Rpi also stays in 1080p mode to do 1080i. Basically, the rpi stays in its mode (no need to reboot), and the DMA engine inside the NeTV2 adjusts to try and do something sensible with the rpi's video data.

xer0design, 2019-04-23:
So, I'm spending the night on this; I've been the past hour trying to get anything at all to show up but I finally got past that, think it was bad HDMI cable - now my issue is that in any mode (1080p, 1080i, 720p), the overlay just shows up as static and appears to be duplicating everything twice. I think it might still be another bad hdmi cable so I'll see if I have anything better laying around but at least with 1080i and 720p it's at least passing through and showing something so I'll see what I can figure out and update again.

bunnie, 2019-04-23:
Sorry to hear about the troubles, thanks for trying it out. 

Hmm, are you entering the firmware command on flterm to switch to the respective mode (e.g. typing 720p or 1080i into flterm)? You shouldn't be seeing the twice-duplication, that would indicate that the NeTV2 didn't actually switch into the correct mode and it's thinking it's still in 1080p mode. And this is, of course, assuming you ran the testing update in the correct branch (multires)....

The NeTV2 will "stick" into the mode you set it into until you reboot it or restart the FPGA. So if you're getting the static, you can unplug the video source and plug it back in again, usually on the second try that problem will get solved.

bunnie, 2019-04-23:
Yikes, i just saw how i left the wrong branch in the multires docs. Thanks for fixing that.

xer0design, 2019-04-23:
Figured out my issue, I think, at least it's working somewhat now. Turns out that the _output_ cable also has to be short and high quality. I was testing there with a fairly long (~10m) run around the building so I could keep the netv close to me and that was the source of my static and duplication issues it seems. 

My current situation is it's working in all three modes, only with a green cast across the whole image (which isn't fixed by setting chromamode0/1), but it's correctly passing through both feeds and overlaying them. I'm going to guess it's something small so I'll keep trying to debug and figure out what exactly is wrong.

Interestingly, when switching the input between resolutions it seems to pick them up a good amount of the time without having to go into filterm, 1080p/1080i seems to work 100% of the time with absolutely no problems

xer0design, 2019-04-23:
Actually, scratch that, it's not working as well as I thought it was as I just had the static desktop displayed and it looked okay; when opening anything it falls apart again, see attached pic, the overlay is actually strobing, kind of looks like it's changing brightness between every second line as if it was some kind of interlacing thing? (it's 720p input with the 720p setting). I'll keep on debugging

[image]

xer0design, 2019-04-23:
Okay, ignore my last post - after plugging out and in the input cable about 5 times it just started magically working perfectly! Phew. Now it seems like everything has been set to saturation -40% but otherwise, it looks like good progress is being made

xer0design, 2019-04-23:
Update: plugging it out and in a few more times fixed the saturation issue. I think I'll go buy some overpriced hdmi cables tomorrow to see if it negates any of these signal issues but I'm happy to say it's working in 720p mode pretty much perfectly - thanks for your help.

1080i is another story. I've tried a few things here but earlier when I was having a bad output, 1080p/1080i worked identically - both gave the same bad output. Now that I have it working as it should look, 1080p is passed through perfectly (even with 1080i set in filterm), and 1080i shows everything in green and pink and stretched and you can only see every second line from the pi. Not too big of a deal, 720p will do me just fine but so you know 1080i isn't functional. 

My 720p issue is with the top left section of the pi, is it possible to make 720p mode work with the pi set to 720p also, so there's no need to make everything fit in the corner and go spending hours trying to figure out an elegant way of setting that all up? I've tried setting the overscan in config.txt to essensially make the pi 720p but that doesn't seem to do anything at all, sadly.

bunnie, 2019-04-24:
Those are some strange issues...

1) surprised that output length matters, but 10m is pretty long. Could also be indicative of maybe the PLL settings being off on alternate rate modes, like I mentioned before they are complicated and I'm drafting off of some partial work I did months and months ago that I found in the archives. However, this is not the primary issue, let's look into that once the bigger fish are caught.

2) It's really strange that you have a saturation issue. I wouldn't expect that at all. 

3) Interesting that chroma option didn't work with the different resolution. Those are two independent state machines. Another thing to look into.

4) When you're in 1080i mode, can you please type
"debug inter"
It should respond with "even pos: / odd pos: " numbers. Can you please report those?

5) When it's struggling to connect -- is there any way you can copy/paste output of flterm to me? Basically, just a log of from when you plugged in the cable to when (if ever) it connects along with the settings you thought you gave it (including the resolution mode you entered into flterm, and the resolution of the source). Ideally, the procedure would look a bit like this:

* run 'pm2 stop netv2-status' to stop the json status update reader from competing with flterm
* Boot the system, connect to flterm
* type 'json off' to stop the json spew
* Set your resolution mode (1080i/720p/1080p)
* run "debug input0" -- this should create a rapid stream of debug spew
* unplug the input cable, and plug it back in again
* from the point where you set resolution mode (including the cable plug) to where the signal syncs (if ever, or about one minute if it doesn't) is the log I'm interested in. This log will help me understand what the exact issue is with your system.

6) as for 720p on the pi, as you noted, a key challenge is getting the pi to be forced into 720p mode. I think I have socked away somewhere an EDID that can trick the pi into 720p mode but you have to reboot the pi to apply it; and to get out of it you have to reboot the pi again. It's...not pleasant. Before embarking on a venture to change the pi's behavior, for your application is it the case where like...once you're in 720p, you're never, ever going to want to go to a different resolution mode? If resolution mode changing dynamically based on e.g. content source preference is even remotely in the cards then I think rebooting the pi is out of the question.

I think for now, maybe let's try and figure out why 1080i isn't working, perhaps once you can send me your debug log as requested in item #5?

Thanks for helping with this. I will mention now, it might be a bit of a long road before this is figured out, at least until I can reproduce the issue you are seeing in my lab, and I also am ping-ponging between this and other projects/travel. But I appreciate your patience!

xer0design, 2019-05-20:
Hey!

Sorry for the delay, life got in the way but I'm back at it for the night. I'll just reply to each point below

_1. Cable length_
Not the main issue, overpriced "oxygen free gold plated" cables seem to have solved the issues. I swapped both the input and the output cables just to be sure.

_2. Saturation_
I haven't seen this issue popup again since, will update if I see it again, though.

_3. Chroma_
This is still very much an issue in 1080i, setting `chromamode` to 0 or 1 has no effect, even after reboot.

_4. "debug inter" for 1080i_
Again, new cables seem to have fixed a lot of the issues apart from the pink/green chroma issues. Not that these seem more severe than they originally were also? I'm not sure but everything is VERY pink and fine lines have a green 1px ghost is how I'd describe it.
Here it is, anyway.
```Disabling json status
RUNTIME>debug inter
even pos: 1099
odd pos: 2199
RUNTIME>
RUNTIME>
```

_5. Connection Issues_
Since swapping out to the expensive cables, the struggling to connect problems seem to have disappeared.

_6. 720p_
If we can solve this chroma issue then 1080i should work just fine in theory, however, if 720p is the way to go forward then yes it will never have to change once set, it'll be 720p forever with no variable input.

xer0design, 2019-05-20:
So, after a few hours of messing about, it's all somehow working near perfectly at 1080i! I don't even remember half of the things I've done so I'll be imaging my SD card as it is in case it breaks again. My main issue now is that everything seems pretty pixilated in the 1080i mode, I'm not sure why but nothing is really as sharp as you'd expect it to be. Take a look at the attached photos to see what I mean; I'm guessing it's something to do with the pixel maps you were talking about earlier? It's all functional just... jagged. 

I've tried with different chroma values for black and pink and green to see if it was a problem with separating the colours but that didn't do anything, and as you can see in file manager where there's no separation taking place the problem is still there. 

The underscan is also a bit annoying but that's a smaller problem for another time. Thanks for your help and patience with my complicated and probably near one-off use case, I appreciate the help!

[image]
[image]
[image]
[image]

bunnie, 2019-05-21:
It's great that you were able to find a combo of settings that gets it to work! If you don't mind sharing the image some day, I think it'd be interesting to see what those are.

As for the grainy text, I'm struggling to find a hypothesis to explain this conclusively. Here's the candidates:

- Interlace fields are swapped (even/odd swap). In this case, on a diagonal line, you would see an "every other line sticking out" phenomenon. This is not the case.
- Interlace fields being duplicated. In this case, aspect ratio is roughly preserved but you'll see that a true diagonal line (as you see on the "Run" paper plane icon) would end up with a 2x1 vertical rectangle instead of what you're seeing there, which is a series of dots.

So it's probably not progressive->interlace problem. The remaining hypotheses I have are not really fixable, but they could explain it:

- YCrCb color space conversion loss. So the whole processing pipeline is the Pi natively renders YCrCb, there's a color space conversion to RGB, then inside the NeTV2 it gets converted to YCrCb again, then the TV converts YCrCb to RGB. There's three round trips happening of color conversion and each one of them will incur some rounding loss. So one possibility is e.g. some shades of gray are clipping out due to rounding errors from the total color conversion pipe. Not much can be done about that
- The TV's deinterlacer is trying to do a smoothing effect to improve perceptual resolution, and that is hurting computer graphics. I think this might be the most likely explanation -- interlace is a little funky, because images are actually progressive. So TVs do some digital signal processing to optimize for real images, e.g. smooth tones and continuous shades. These algorithms trade off integrity of sharp computer generated graphics (single-width lines, like you see in the fonts) for better perceptual quality of camera images. Again, this might be something you can tweak on the TV side but I'm not sure.

The artifacts you're showing look like something that requires a DSP kernel to generate. The NeTV2 itself does virtually no signal processing aside from color space conversion, so it shouldn't care if a pixel is part of an anti-aliased line or not...

xer0design, 2019-05-21:
It's not a TV issue, now that 1080i is working I've fully installed the box while I continue development so it's coming from a stb, going into a hdmi matrix, and outputting to 10 TVs - all different models and sizes, all exhibiting the same effect - it's not related to the matrix either, I tried direct connection to a TV and its the same. Graphics generated by the stb don't exhibit the same jaggedness either. Should I maybe try different output settings in config.txt? I tried 1080p and 1080i but both look the same.

Thinking out loud, maybe the netv overscanning the pis source is causing some problem? There's a I'd guess 5% border around the pi image where nothing is being overlaid

xer0design, 2019-05-21:
Okay I'm back again for the night. Set everything back over to 720p to see and that jagged issue is completely resolved, everything is absolutely crystal clear - so for now, would it be complicated for you to tell me how to set the pi to 720p also? As said it'll be locked into that mode, and 1080i can come down the road, but I'd really like to keep the progress going for the moment so 720p on the pi is the easiest way I can see for going about that, unless you think the jagged issue has a fix

bunnie, 2019-05-21:
Unfortunately setting the Pi to 720p also means having to change the input capture on NeTV2 to match that, as well as editing the EDID presented by the NeTV2 (which is compiled into the FPGA) -- it's actually not straightforward. You can't do the thing where you scale all the images to a window in the top left corner in a 1080p window?

xer0design, 2019-05-21:
I was afraid of that, I understand. I'm sure I can, I'm generally calling things fullscreen from the command line (cvlc --fullscreen, kweb --kiosk, etc) to make everything just load and fit automatically without any hassle. For video, I suppose I can reformat it so the content stays in the corner and for webpages set them all to iframe in the corner maybe. Thanks anyway! Do you think the interlacing problems are similarly complicated to resolve?

bunnie, 2019-05-21:
I think ... the interlacing problem, if I can reproduce it, may be easier to resolve and less hairy from a system integration standpoint. Until I can find a way to get a Pi to switch resolutions without modifying config files and a reboot the 720p rendering from the Pi is going to be problematic I /think/. Maybe I haven't thought hard enough about it yet but just getting the Pi to snap into 1080p instead of some weirdo default "safe" resolution with our synthetic EDIDs was an "ok it works now I don't know why stop touching it" situation. Unfortunately the docs on the Pi-to-HDMI back end are /really/ thin.

At the very least, I'd like to give the 1080i issue a try to reproduce before trying the 720p problem.

xer0design, 2019-05-21:
No worries, let me know if you need anything else from me from a code of debug standpoint and I'll be sure to drop it in here! Thanks

bunnie, 2019-05-24:
OK, I found the bug. 

e3514be72f00734ebcd827a69a4d3510702b7048

Fixes the low-res issue in 1080i. I was not detecting the even/odd field correctly, so it was just doubling one of the lines instead of grabbing alternating fields, hence the low resolution. 

I've tested it through a few reboots on my side and it seems to pick up the correct interlace polarity each time. Please give it a try and let me know if this solves the problem for you.

xer0design, 2019-05-24:
You beautiful man, thank you! I'll give this a go as soon as I can and report back! Just pull the code and run testing-fpga.sh I assume?

bunnie, 2019-05-24:
yes, that's correct, ideally, that should be all you need to do.

and of course, type "1080i" into flterm, to select the correct mode.

plaes, 2019-06-05:
Tested it a bit with 1080i, mostly works ok but noticed bunch two issues:
1) Set 1080i mode, reboot and the NeTV2 status overlay is sometimes shifted ~30 pixels to the left (so half of H is missing). Setting the mode again using `1080i` sometimes corrects this.

2) Serial terminal (flterm) starts losing incoming data:
```
RUNTIME>status
inpu0:192040 (  7. MHz)
input1:  1901080 ( 849 MHz
adc: 747 mC
dr: rad: 176bps  wite 3986b  all:772Mbp
```

bunnie, 2019-06-05:
flterm is losing data because you need to stop the netv2_status service, it's stealing characters.

Please run 

pm2 stop netv2-status

before running flterm to avoid that problem.

plaes, 2019-06-05:
Thanks, @bunnie, one issue already fixed, but found another:

[image]

bunnie, 2019-06-05:
You've using the chroma on path, yes? I think unfortunately with chroma on, there's a line-wrap issue that happens due to the latency of the reformatting. It would be best if you can just avoid using the far left side for overlays, but if you must use that portion of the image open a separate issue. 

It's a tricky one to fix, it might take a while....

plaes, 2019-06-05:
Don't really know what I'm using.. as I just received the netv2 today and tried out the multires branch and haven't actually changed anything else.

bunnie, 2019-06-05:
OK. If you can document the settings you're using and open a separate issue that'd be helpful for me to reproduce and isolate. Sorry for the troubles on multires, but it is experimental...

bunnie, 2019-09-22:
closing this bug for now, as the latest release /should/ address 1080i automatic detection and switching.

closed 2019-09-22 by bunnie