AlphamaxMedia/netv2-fpga #8: README isn't clear on key points

tommythorn, 2019-04-04:


bunnie, 2019-04-04:
I had made a wiki page for this, but failed to link to it in the Readme.

https://github.com/AlphamaxMedia/netv2-fpga/wiki/Using-NeTV2-as-a-Dev-Board

you are also right, I should probably explain more on how to prepare your own raspberry pi image. I've got an image uploaded and a burner but its documented from the perspective of a factory looking to churn out hundreds of images. I'll try to cobble together something more appropriate for individuals in the coming week.

bunnie, 2019-04-04:
in case you're curious here is the image burning tool:

https://github.com/AlphamaxMedia/usb-pyromaniac/blob/master/README.md

and the production image partitions are at

https://bunniefoo.com/netv2/production/

tommythorn, 2019-04-04:
Thanks Bunnie, but all my comments actually referred to https://github.com/AlphamaxMedia/netv2-fpga/wiki/Using-NeTV2-as-a-Dev-Board (shouldn't have written "README", sorry).

I don't see an answer to 1.  How exactly are we reaching JTAG on the FPGA?

bunnie, 2019-04-04:
Sorry, you're right, probably a bunch of others will share your experience.

The purpose of the stand-alone board is so that people can build a full "NeTV2" system without having to pay the mark-up for me to provision a Raspberry Pi 3B+. Because of the costs involved at the factory and in holding the inventory, individuals who want to "DIY" an NeTV2 can save a bundle by building their own system.

I hadn't really thought a lot about the "I wanted just a stand-alone board" use case scenario -- my assumption is people who bought the board only were doing it for the purpose of cost savings over the bundled kit, vs they just wanted to dev on the NeTV2 as an FPGA development board. In retrospect that seems obvious, but during the campaign the logic went that if you wanted an  Artix dev board you'd probably already own an Arty or something like that, and thus the main reason you'd get an NeTV2 was specifically for the NeTV2 use scenarios , which is pretty intricately intertwined with  host computer like an Rpi.

But answering your question more directly, yes, the NeTV2 was developed specifically in mind with the Raspberry Pi 3B+ serving as an SSH-to-JTAG dongle, and so the NeTV2 was developed with a 5V power supply with sufficient amperage such that it could power an Rpi slaved off the main board plus the primary application without a glitch -- the Rpi support is viewed as more of a "you're saving the cost of a JTAG to USB adapter", and once you've developed your application you'd burn it into the SPINOR on the main board and only then you can detach the Raspberry Pi and run from the DC wall adapter without any additional support.

I'll try to build a doc path from the perspective of those who bought the board with no intention to use the board for the NeTV2 application space to help address your concerns, but it might be a couple weeks out before I can get to it.

tommythorn, 2019-04-04:
Thanks.  I appreciate that I don't represent the target audience of the board and requiring that it be attached to a Raspberry Pi is fine.  It just wasn't clear to me what was assumed.

However it leaves me which an new question: what are the plans for PCIe?  I'll assume that it's secondary and not fully developed yet, but in a scenario where the NeTV2 is plugged into a motherboard, is the Raspberry Pi still assumed to be attached?

I apologize if this isn't the preferred way to asking these questions.

bunnie, 2019-04-05:
PCIe support is being developed by the user community. Most of the PCIe work is being done by enjoy-digital, he has a litex core for PCI express:

https://github.com/enjoy-digital/litepcie

My understanding is that it's been lightly tested on NeTV2 but there hasn't been a lot of bandwidth to do a full port of the core to the NeTV2 system. You are correct in assuming that I want to focus 100% of my attention on supporting the "out of box" use case, there are a number of users who want to integrate the stand-alone solution and my primary concern for now is to support them. Once the initial launch bugs have been shaken out, perhaps I will be able to also jump in on the PCIe port, but if you're really interested in it, I'd encourage you to check out the pcie core and give it a try. 

As whether the Rpi must be attached or not, it's mainly a matter of if you want to have access to the UART and JTAG. The system has an on-board SPINOR, so it's capable of booting directly to the PCI express application, and you can also update the SPINOR from PCI express as well. So it's theoretically possible to do all development without an attached RPi. But I personally find life to be a lot easier when I have JTAG and UART access.

bunnie, 2019-04-17:
I've added a link to the README.adoc file for bare board users, and am fleshing out that part of the wiki now. Hopefully this resolve the issue.

closed 2019-04-17 by bunnie

tommythorn, 2019-04-17:
Thanks, this is very helpful.