AlphamaxMedia/netv2-fpga #9: Documentation?

tommythorn, 2019-05-21:
The docs states "The latest hardware docs are maintained in a separate git repository." but doesn't name any or point anywhere. It's doesn't look like any of the AlphamaxMedia repos are the documentation for this.

bunnie, 2019-05-21:
Can you point me to where you're seeing this? If you go to 

https://github.com/AlphamaxMedia/netv2-fpga/wiki

You'll find that "PCB, plastics, production testing software" are each hyperlinks to three separate repositories for the PCB, plastics, and production testing software respectively.

tommythorn, 2019-05-21:
https://github.com/AlphamaxMedia/netv2-fpga/tree/master/doc

bunnie, 2019-05-21:
OK, that's an out of date stub and I've deleted it. Please refer to the README file for the pointer to the wiki which contains the links I mentioned above to all the hardware docs.

closed 2019-05-21 by bunnie