AlphamaxMedia/netv2-fpga #11: [experimental-multires] Issues with 1080i

plaes, 2019-06-05:
When testing multiarch branch with 1080i I ran into following issues:

1) One pixel shift of lines
[image]

2) Garbled overlay colors (seems to always happen with one certain TV channel)
[image]

3) Whole overlay shifted left after rebooting FPGA (~3 times out of 10 reboots). Sometimes reinitializing with `1080i` command helps.

[image]

bunnie, 2019-06-06:
Thanks for reporting these. this is actually three separate issues. Since they are all grouped together I will just cherry pick the most convenient one to work on and branch it to a new issue when I get the time.

bunnie, 2019-09-22:
I think the most recent patch pushed should address the 1st and 3rd issues. The 1st issue for sure is fixed. The 3rd issue you might still see about an 8-pixel shift from boot to boot, but I haven't been able to reliably reproduce the problem. I think it's related to the fifo sync between the overlay and oncoming stream, and it seems rare enough I'll close this for now and we can open again in a separate bug if it's really an issue.

The color being off on a particular channel -- I haven't been able to reproduce that problem, but also, if the problem is due to color space being changed on a given channel by the broadcaster, there's nothing I can actually do about that because the colorspace info is encrypted and it would be a 1201 DMCA violation to read that out and adjust the overlay to match. So that bug may have to wait for the courts to decide this is a thing we should be able to do with our video.

closed 2019-09-22 by bunnie