AlphamaxMedia/netv2-fpga #13: Fuzzy overlay text

peteboz, 2019-10-26:
Hi.  

1. I just got my netv2/pi combo and I'm seeing fuzzy text on the overlay.  My source is a mac mini, I've updated the firmware to the latest and changed the colorspace to ycrcb since I was getting pink and green text.  I've also tried swapping cables and re-plugging cables.  Please see screenshots below.
2. Also, for some reason the overlay "block" is clear instead of black, how do we change this?
3. What's the best way to shut down the system?  I assume pulling the plug isn't good for the pi.

Thanks in advance!

[image]
[image]
[image]

peteboz, 2019-10-26:
Just a quick update.  I did an experiment and brought up a completely black screen for the overlay and it looks nice and sharp now...  The fuzziness seems to appear on non-dark sections of the picture.  Is this normal?

I guess my other 2 questions still remain:

2. For the reasons above, how do we change the overlay "box" this to get a black box effect, since the text is a little difficult to see on lighter backgrounds?
3. What's the best way to shut down the system? I assume pulling the plug isn't good for the pi.

Thanks again!
[image]

xer0design, 2019-10-26:
Hey! 

The pink/green thing can be fixed by pulling the ycbcr branch and changing the chroma mode, check the issues and you’ll see when i brought it up and how to fix it. lm

bunnie, 2019-10-26:
Hi there --

I think I am understanding your issue, but correct me if I'm wrong. 

I believe the "fuzziness" is due to anti-aliasing of the text. The text edges anti-alias to black, and the anti-aliasing is being chroma keyed as well. One option you have is to change the chroma key range, so that you're not compositing darker shades of gray onto your video (see https://github.com/AlphamaxMedia/netv2-fpga/wiki/Customizing-Chroma-and-Colorspace for commands).

The "black box" effect -- again, this would be done by futzing with the chroma. Right now, black *is* the key color, so black by default is replaced with video. One trick you can use is just do a really dark gray box that is just lighter than the chroma threshold. The other trick is you can pick a color like "magic pink" to replace -- basically, a section of the gamut that tends not to find its way into your overlay UI. If you make the key "magic pink" then only areas that are that color in the raspberry pi's UI will be replaced by video, allowing black to be composited.

Shutting down the pi -- if you have the wherewithal to log in, best way is to just do a sudo shutdown -h now and then wait until the pi shuts down and then you can pull power. That would be the cleanest way to do it.

peteboz, 2019-10-27:
Hi bunnie - thank you for the reply! For the "fuzziness" issue, i think you are right on the anti-aliasing.  It's most obvious on the news feed module as it transitions from one story to the next.  

I tried the steps for "Changing overlay thresholds for YCrCb" on the wiki page you referenced, but this just made the entire screen black (no video) with the overlay shown.  Was I supposed to change the values to something other than:
chromalo 0xeb00eb
chromahi 0xff10ff
chromapol 1

I changed the background css color in the Magic Mirror css to a dark grey and that worked to create a black box effect, but I think I like the default transparent look better if I can solve the anti-alias issue and make the text more visible.

Thanks again!

bunnie, 2019-10-28:
Ah sorry, those settings are for YCrCb mode. If you're in RGB mode, try doing

chromahi 0xffffff
chromalo 0x808080
chromapol 0 

This basically says, "Colors rendered by the raspberry pi between RGB #808080 and RGB #FFFFFF should be overlaid on the video feed". The default is RGB#141414, by changing the chromalo to #808080 it should reduce the black outline due to aliasing, but it will make your text look a little less nice.

Unfortunately, the "right" solution -- which is to alpha blend the text -- is not an option because there is a risk that section 1201 of the DMCA makes it unlawful to decrypt the background video even for the purposes of alpha blending.

peteboz, 2019-10-28:
Hi bunnie, thanks.  Strange - the settings don't seem to make a difference either way.  Am I supposed to be on the testing-fpga version?  I'm on the latest.

Thanks!

bunnie, 2019-10-28:
You should be using the latest FPGA pull that's checked into master -- all of these commands got "promoted" into the master build recently. 

That is strange. If you flip chromapol to 1, that should *really* change things. If that changes nothing then maybe the right version isn't on the FPGA, or I borked something else.