AlphamaxMedia/netv2-fpga #14: Problem with adjusting cropping

Lanthade, 2019-11-18:
I've been trying to adjust the cropping per the posted instructions and every time I try it I completely loose the overlay.

I'm not sure if the command I should be issuing is:

`debug setrect 32 32 10 10
`

or if it should be:

`debug setrect <32> <32> <10> <10>
`

Either way I loose overlay till I do a reboot on the fpga board (thanks for the button BTW).

bunnie, 2019-11-18:
The cropping got phased out in the latest firmware release. Have you
updated your firmware since you received the board?

The original purpose of the cropping was to prevent image material from
smashing sync signals that would cause you to lose the overlay. With the
latest release the sync areas are detected and automatically masked out,
so if there is some problem aligning the pixel streams you don't
accidentally smash the sync signals on the outgoing stream.

The idea being that subsequent to this, you can the crop the image
reliably by rendering it within a box, instead of having to rely on this
feature (which was unreliable and can cause the sync loss as you noted).

Lanthade, 2019-11-18:
Ahh, good to know.  Maybe mark that instruction page as depreciated then?

bunnie, 2019-11-19:
Good point. I've gone and added a note to the page.

closed 2019-11-19 by bunnie