AlphamaxMedia/netv2-fpga #16: Making changes sticky across reboots

Lanthade, 2019-11-18:
Is there a way to make FPGA changes sticky across reboots/power cycles?  For instance I always have to manually set color space to ycrcb on each power up.

bunnie, 2019-11-18:
I'd recommend using a startup script to run the command, e.g. something
similar to

echo "your custom command" > /dev/ttyS0

inside a .sh file that is sequenced by systemd to run a bit after the
power-on.

The only possible gotcha is if /dev/ttyS0 is not configured for 115200
baud, if that seems to be the case I can help search for the incantation
to set the baud rate before you do that.

Lanthade, 2019-11-18:
Got it.  I'd vote for a feature request to accomplish this on the FPGA board if that's possible in the future.  It just strikes me as more reliable.  Thanks!

bunnie, 2019-11-19:
That's a possibility. I don't think write capabilities are built into the litex SPI block that the FPGA is using, so it might be hard to burn in, but I'll leave this issue open so I can keep that in mind for the next dev cycle.

mithro, 2019-11-19:
@bunnie - LiteX BIOS has a demo on how to write to the SPI flash -> https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/spiflash.c 

https://github.com/enjoy-digital/litex/blob/d905521185fada6fe246e546384eb19e6188e030/litex/soc/software/bios/main.c#L181-L221
```c
#if (defined CSR_SPIFLASH_BASE && defined SPIFLASH_PAGE_SIZE)
static void fw(char *addr, char *value, char *count)
{
	char *c;
	unsigned int addr2;
	unsigned int value2;
	unsigned int count2;
	unsigned int i;

	if((*addr == 0) || (*value == 0)) {
		printf("fw <offset> <value> [count]\n");
		return;
	}
	addr2 = strtoul(addr, &c, 0);
	if(*c != 0) {
		printf("incorrect offset\n");
		return;
	}
	value2 = strtoul(value, &c, 0);
	if(*c != 0) {
		printf("incorrect value\n");
		return;
	}
	if(*count == 0) {
		count2 = 1;
	} else {
		count2 = strtoul(count, &c, 0);
		if(*c != 0) {
			printf("incorrect count\n");
			return;
		}
	}
	for (i=0;i<count2;i++) write_to_flash(addr2 + i * 4, (unsigned char *)&value2, 4);
}

static void fe(void)
{
	erase_flash();
	printf("flash erased\n");
}
#endif
```