AlphamaxMedia/netv2-fpga #17: Overlay not showing

cluppi, 2020-03-01:
Just got a netv2 quickstart and I can't seem to get the overlay to render.
I have a Windows machine as input source. Tried two different monitors (a 24 and a 21 inch, both 1080p@60). I can see the windows source just fine but no overlay.
Tried ssh'ing into the device, updated fgpa, restarted MM and everything looks good but no overlay. Also tried running GUI apps attached to the display :0 but nothing there either.
Got a black screen when trying `sudo openocd -f ./fpga-jtag.cfg`, which went away after a restart.

Any ideas?

bunnie, 2020-03-02:
Have you tried using a cable box or a DVD player as a source? I have found laptops to be a bit tricky because they can output weird sync signal combinations. My best guess is that maybe it has either invented sync polarity or some non-standard timing that's throwing off the overlay engine's genlock algorithm. The problem being that a lot of monitors will offer the kitchen sink in terms of supported resolutions and the laptop will pick a weird one because it's alphabetically the first in the list or something like that, while consumer electronics usually just defaults to the least common denominator because it's "safest" in terms of a compatibility standpoint.

If you can try plugging into a "dumb" consumer electronics device and confirm that an overlay appears then that can narrow down the possibilities for what's going wrong...

cluppi, 2020-03-02:
That was very helpful and I appreciate the quick response. I tried combining pretty much all inputs and outputs I have here and this is what I found:

### Combinations that work:
Roku TV ->  Samsung 50" TV (un50eh6000f)
Linux / GeForce GTX 960 -> Samsung 50" TV (un50eh6000f)

### Problem rendering overlay / static:
Windows Desktop / Radeon RX 5700 XT -> Samsung 50" TV (un50eh6000f)

### No changes to input signal / no overlay:
Windows Desktop / Radeon RX 5700 XT -> (Acer G246HL 24-Inch monitor)
Linux / GeForce GTX 960 -> (Acer G246HL 24-Inch monitor)

--

I'm using a DVI -> HDMI cable from my monitors to PC / netv2. You think that using the VGA port might force some sort of backwards compatibility setting and get the overlay to be shown? I can try buying an adaptor if you think that might work.

bunnie, 2020-03-02:
ah, you are using DVI. it actually has a material coding difference from HDMI. you can try typing

debug dvimode0

into the serial console to the fpga via flterm. this should switch the mode of the rx0 port to DVI mode. I've never really tested this option, though, so ymmv. The setting is volatile so next boot it's back to HDMI mode. 

https://github.com/AlphamaxMedia/netv2-fpga/blob/936d239bc2d2c93c0fc979edec929ea728a391d4/firmware/ci.c#L1117

alternatively, you can try an rgb converter, it may output HDMI instead of DVI.

cluppi, 2020-03-02:
ah that did fix the issue when testing Linux / GeForce GTX 960 -> (Acer G246HL 24-Inch monitor). That works perfectly.
I could also see the overlay with static/snow when testing Windows Desktop / Radeon RX 5700 XT -> (Acer G246HL 24-Inch monitor). This is ideally the setup I want this to work for. Do you think I'm getting static because of the OS or the GPU? I can try to swap the GPUs later tonight to see if I can get windows to render the overlay without static
Thanks again for your help!

cluppi, 2020-03-02:
Ah never mind! I found a setting called "HDCP Support" (High-bandwidth digital content protection standard) in my GPU software and figured that was what could be messing with the encryption. Disabling that makes it work for my windows/radeon setup 100%.
Everything looks good and this device is amazing, good work!

bunnie, 2020-03-03:
thanks, glad you got it working!

closed 2020-03-03 by bunnie