AlphamaxMedia/netv2-fpga #18: Only a black screen at NeTV2 output no matter what input or output

aidanmoore, 2020-03-17:
Hi there,

I have purchased a "just the board" NETV2 and I've been reviewing the notes here and for the last week trying different combinations of sources, sinks and cables with absolutely no video output with any combo.

So, here is what I know works:

- Downloaded RPi image and flashed no issues, magic mirror starts and I can access the other menu items.
- FPGA update works (in that it does not throw an error while uploading) . 
- Since I didn't purchase the custom HDMI overlay cable, I also uploaded test FPGA code for standard HDMI cable on overlay port and it uploaded correctly.
- The RPi is operational , connected via ethernet and I can see the Magic Mirror Interface on the RPi HDMI output.

Understood about sensitivity of Rx HDMI input to short cable and stable source. I've confirmed 1080p60 output of the signal source into the NeTV2 using a Atomos Shogun monitor in loop through, and it reports 1080p60 on three sources I have used as the Main input:
Laptop 1 1080p60
Laptop 2 1080p 60
Blackmagic 4K Cinema camera

I also read about trying an encrypted 1080p source, so I hooked the NETV2 up to my cable TV box (1080p) and my Samsung 50' HDMP TV. Still not a hint of any video activity at the ouptut. 

Other notes:

1. I have also tried 1080p30 and still no output.
2. The DRAM seems very hot: 93.5C on a FLIR readout so I added a heatsink.
3. I have tried to move the jumpers between the default "snoop" and "source" settings with all three sources above.
4. Confirmed waiting 2-5 minutes on each reboot to see if video shows up.

The board status is always like this:
Solid Green LED near overlay HDMI port
Flashing Green LED near HDMI output port.
LED off at HDMI main input port
I also tried the alternate HDMI output port for activity in all the above configurations.

So, as you see I'm kind of stuck as to what to try next, and your help is greatly appreciated.

Thanks

Aidan

BTW: I inadvertently posted a few versions of the above notes to the subreddit site last week thinking that was tech support. Please ignore those posts.

bunnie, 2020-03-17:
Hi Sorry -- I don't know what is going on with Reddit. I only use it for NeTV2 support and I don't see /any/ new messages there. I must not be looking in the right spot or something weird is going on. Maybe I will replace it with a note redirecting to github, because I actually get notifications from this medium.

I'm trying to figure out the exact configuration that you have. From what I read it looks like:

Cable TV box 1080p -> NeTV2 -> Samsung 50' HDMP TV

Can you please confirm that you have the Cable TV box plugged into RX0, and the TV plugged into TX0? 

Also please confirm that you are using the "snoop" mode for the jumper configuration (both jumpers on snoop). 

Even without an overlay, it should at least pass through video. The lack of a green light near the RX0 input indicates no PLL lock. 

Are you able to open an `flterm` console to talk to the FPGA via serial? If you can, we can try issuing some debug commands to see what the RX0 state machine is thinking.

aidanmoore, 2020-03-17:
Thanks for the uberspeed response. 

Confirming : Cable TV box 1080p -> NeTV2 -> Samsung 50' HDMP TV
and the jumpers in the "snoop" position.

Just connected an Ethernet cable directly to the NeTV2 ethernet port. I cannot see a new DHCP address..is that how a console connects to the actual NETV2? I have a seperate  Ethernet connection to the RPi.

Is there a link with instructions on how to access the FPGA via console?

Thx A

bunnie, 2020-03-17:
The ethernet ports are confusing. 

The ethernet on the FPGA is only for a protocol known as "etherbone", which is mostly going to be used by people developing custom Litex firmwmares. 

The ethernet port you want to plug into is the RPi. The overall topology goes:

Your computer -> Rpi (via ethernet or wifi) -> GPIO header (via serial) -> FPGA -> "softcore" RISC-V CPU inside the FPGA

Assuming you can get shell on the Raspberry pi, this is how you engage the FPGA:

```

pm2 stop netv2-status
cd ~/code/flterm
./flterm --port /dev/ttyS0 --speed 115200
```

I'm assuming you're using the stock NeTV2 firmware, if you don't have flterm you'll need to grab it and compile it from https://github.com/timvideos/flterm

Once you're in, you should see something akin to a terminal prompt. You'll have some spew coming out on the console, to turn it off type `json off`

Now that things are quiet, type

`debug input0`

This will create a pretty fast stream of spew, which can be a little hard to take in. What's most helpful is to see what the spew says when the cable is not plugged in, and then once you plug in the cable, how it changes. The output from that should be able to reveal the issue with the RX0 line.

aidanmoore, 2020-03-17:
Ok...all went as per instructions until I run flterm. 

At that point I get a BIOS> prompt and the remaining commands are not recognized. 

Old school screengrab attached as jpeg.

[image]

bunnie, 2020-03-18:
OK, that's a problem. If it's stuck at BIOS it's unable to load the firmware to boot, probably related to the hot RAM issue. I wonder how the unit passed test at the factory...sorry for the inconvenience, anyways. I will need to investigate this.

Can I arrange for an exchange of your board? Please email me at alphamaxmedia.com, to the user 'support', so that we can exchange your contact info.

johnknaack, 2020-07-08:
I'm getting the same output as the users above:
`pi@netv2mvp:~/code/flterm $ ./flterm --port /dev/ttyS0 --speed 115200
[FLTERM] Starting...
    
BIOS> json off
Command not found
BIOS> debug input0
Command not found
BIOS> `

Is there a way to fix this? I've also email support at alphamaxmedia.com, any other solutions?