AlphamaxMedia/netv2-fpga #19: No output. netv2-status stuck in reboot loop

slack69, 2020-04-06:
I've just bought the NETV2 from crowdsupply in it's quick start enclosure. The output port is constantly flashing and the input port does not flash at all. It does not seem to matter whether I have hdmi cable plugged in. I've tried 3 different sources and 2 different outputs, all to no avail.

After sshing in I can see netv2-status has a large number of restarts, I stopped it and ran `netv2-status.sh` which gave me this error:
[image]

I've tried running `update-fpga.sh` which appeared to complete successfully.

Could someone please advise what to try?

slack69, 2020-04-06:
Changed line 39 to `var status={};` which seems to be the issue with the error as status is empty.
I added back in the `debug console.log()` and it only appears to be writing the message to the serial port. Not erroring or receiving anything.

bunnie, 2020-04-06:
The output port flashing just means the FPGA configured correctly. 

Input port not flashing means there was no PLL lock on the input video. 

Can you run flterm and let me know what you see:

```
pm2 stop netv2-status
cd ~/code/flterm
./flterm --port /dev/ttyS0 --speed 115200
```

Once you're in flterm you'll probably see some debug spew; type

`json off`

and that spew should stop

Then run

`debug input0`

and plug in the cable, and let me know what you see...

slack69, 2020-04-06:
Seems to be stuck starting. No seeing any debug spew.
[image]
[image]

slack69, 2020-04-06:
Tried pulling the latest updates to flterm and rebuilding but I get the same output.

bunnie, 2020-04-06:
Humm ok. Can you open a second terminal, so that you haven't closed the flterm terminal, and on the second terminal do the following:

```
cd code/netv2mvp-scripts
sudo openocd -f reboot.cfg
```

When you go back to flterm, you should get some output a bit like this:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
 SoC BIOS / CPU: VexRiscv /  75MHz
(c) Copyright 2012-2018 Enjoy-Digital
(c) Copyright 2007-2018 M-Labs Limited
Built Sep 13 2019 03:04:54

BIOS CRC passed (5705140b)
Die temp: 59.30C
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: 32+-00
m0, b1: |00000000000000000000000000000000| delays: 32+-00
m0, b2: |00000000000000000000000000000000| delays: 32+-00
m0, b3: |00000000000000000000000000000000| delays: 32+-00
m0, b4: |11111111111111111111111100000000| delays: 12+-12
m0, b5: |00000000000000000000000000000000| delays: 32+-00
m0, b6: |00000000000000000000000000000000| delays: 32+-00
m0, b7: |00000000000000000000000000000000| delays: 32+-00
best: m0, b4 delays: 11+-11
```

If you can copy/paste that output here, that would be helpful.

When did you buy your NeTV2 quickstart?

slack69, 2020-04-06:
Here's the output:

```[FLTERM] v2.4-31-g8898c78 Starting...

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
 SoC BIOS / CPU: VexRiscv /  75MHz
(c) Copyright 2012-2018 Enjoy-Digital
(c) Copyright 2007-2018 M-Labs Limited
Built Sep 13 2019 03:04:54

BIOS CRC passed (5705140b)
Die temp: 64.84C
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: 32+-00
m0, b1: |00000000000000000000000000000000| delays: 32+-00
m0, b2: |00000000000000000000000000000000| delays: 32+-00
m0, b3: |00000000000000000000000000000000| delays: 32+-00
m0, b4: |00000000000000000000000000000000| delays: 32+-00
m0, b5: |00000000000000000000000000000000| delays: 32+-00
m0, b6: |00000000000000000000000000000000| delays: 32+-00
m0, b7: |00000000000000000000000000000000| delays: 32+-00
best: m0, b0 delays: 32+-00
m1, b0: |00000000000000000000000000000000| delays: 32+-00
m1, b1: |00000000000000000000000000000000| delays: 32+-00
m1, b2: |00000000000000000000000000000000| delays: 32+-00
m1, b3: |00000000000000000000000000000000| delays: 32+-00
m1, b4: |11111111111111111111111110000000| delays: 12+-12
m1, b5: |00000000000000000000000000000000| delays: 32+-00
m1, b6: |00000000000000000000000000000000| delays: 32+-00
m1, b7: |00000000000000000000000000000000| delays: 32+-00
best: m1, b4 delays: 12+-12
m2, b0: |00000000000000000000000000000000| delays: 32+-00
m2, b1: |00000000000000000000000000000000| delays: 32+-00
m2, b2: |00000000000000000000000000000000| delays: 32+-00
m2, b3: |00000000000000000000000000000000| delays: 32+-00
m2, b4: |11111111111111111111111110000000| delays: 12+-12
m2, b5: |00000000000000000000000000000000| delays: 32+-00
m2, b6: |00000000000000000000000000000000| delays: 32+-00
m2, b7: |00000000000000000000000000000000| delays: 32+-00
best: m2, b4 delays: 12+-12
m3, b0: |00000000000000000000000000000000| delays: 32+-00
m3, b1: |00000000000000000000000000000000| delays: 32+-00
m3, b2: |00000000000000000000000000000000| delays: 32+-00
m3, b3: |00000000000000000000000000000000| delays: 32+-00
m3, b4: |11111111111111111111111111000000| delays: 13+-13
m3, b5: |00000000000000000000000000000000| delays: 32+-00
m3, b6: |00000000000000000000000000000000| delays: 32+-00
m3, b7: |00000000000000000000000000000000| delays: 32+-00
best: m3, b4 delays: 13+-13
SDRAM now under hardware control
Memtest bus failed: 177/256 errors
Memtest addr failed: 8192/8192 errors
Memtest data failed: 522240/524288 errors
Memory initialization failed
```

I bought it on the 30th of march and it arrived today.

slack69, 2020-04-06:
I unplugged my input and got: 
```[FLTERM] v2.4-31-g8898c78 Starting...

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
 SoC BIOS / CPU: VexRiscv /  75MHz
(c) Copyright 2012-2018 Enjoy-Digital
(c) Copyright 2007-2018 M-Labs Limited
Built Sep 13 2019 03:04:54

BIOS CRC passed (5705140b)
Die temp: 64.60C
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: 32+-00
m0, b1: |00000000000000000000000000000000| delays: 32+-00
m0, b2: |00000000000000000000000000000000| delays: 32+-00
m0, b3: |00000000000000000000000000000000| delays: 32+-00
m0, b4: |11111111111111111111111100000000| delays: 11+-11
m0, b5: |00000000000000000000000000000000| delays: 32+-00
m0, b6: |00000000000000000000000000000000| delays: 32+-00
m0, b7: |00000000000000000000000000000000| delays: 32+-00
best: m0, b4 delays: 12+-12
m1, b0: |00000000000000000000000000000000| delays: 32+-00
m1, b1: |00000000000000000000000000000000| delays: 32+-00
m1, b2: |00000000000000000000000000000000| delays: 32+-00
m1, b3: |00000000000000000000000000000000| delays: 32+-00
m1, b4: |11111111111111111111111110000000| delays: 12+-12
m1, b5: |00000000000000000000000000000000| delays: 32+-00
m1, b6: |00000000000000000000000000000000| delays: 32+-00
m1, b7: |00000000000000000000000000000000| delays: 32+-00
best: m1, b4 delays: 12+-12
m2, b0: |00000000000000000000000000000000| delays: 32+-00
m2, b1: |00000000000000000000000000000000| delays: 32+-00
m2, b2: |00000000000000000000000000000000| delays: 32+-00
m2, b3: |00000000000000000000000000000000| delays: 32+-00
m2, b4: |11111111111111111111111110000000| delays: 12+-12
m2, b5: |00000000000000000000000000000000| delays: 32+-00
m2, b6: |00000000000000000000000000000000| delays: 32+-00
m2, b7: |00000000000000000000000000000000| delays: 32+-00
best: m2, b4 delays: 12+-12
m3, b0: |00000000000000000000000000000000| delays: 32+-00
m3, b1: |00000000000000000000000000000000| delays: 32+-00
m3, b2: |00000000000000000000000000000000| delays: 32+-00
m3, b3: |00000000000000000000000000000000| delays: 32+-00
m3, b4: |11111111111111111111111111000000| delays: 13+-13
m3, b5: |00000000000000000000000000000000| delays: 32+-00
m3, b6: |00000000000000000000000000000000| delays: 32+-00
m3, b7: |00000000000000000000000000000000| delays: 32+-00
best: m3, b4 delays: 13+-13
SDRAM now under hardware control
Memtest OK
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from flash...
Loading 131072 bytes from flash...
Executing booted program at 0x40000000

NeTV2 software built Sep 13 2019 03:31:43
setting optimized bandwdith on hdmi in 0
idelay_freq = 300000000
setting up HDMI0 interrupts

selecting delmech = 1
hdmi_in0: setting algo 2 eye time to 14 IDELAY periods
hdmi_in1: setting algo 2 eye time to 14 IDELAY periods
interrupt mask (hdcp): 1c
hdcp_ev_enable_read: 1
RUNTIME>{"hdmi_Rx_hres" : 0, "hdmi_Rx_vres" : 0, "hdmi_Rx_pixel_clock" : 0, "hdmi_Rx_phase" : "0 0 0", "hdmi_Rx_symbol_sync" : 000, "hdmi_Rx_sync_pos" : "0 0 0", "hdmi_Rx_symbol_errors" : "0 0 0", "overlay_hres" : 0, "overlay_vres" : 0, "overaly_pixel_clock" : 330156, "overlay_phase" : "0 0 0", "overlay_symbol_sync" : 111, "overlay_sync_pos" : "0 0 0", "overlay_symbol_errors" : "0 0 0", "ddr_read_Mbps" : 135, "ddr_write_Mbps" : 0, "hdmi_Rx_eye_opening" : "0 0 0", "overlay_eye_opening" : "0 0 0", "fpga_die_temp" : "66C" }```

bunnie, 2020-04-06:
Huh that's weird. And when you plug it back in again, does it stop working?

slack69, 2020-04-06:
Okay really weird it's just started working? I'm sorry for wasting your time I guess :/

bunnie, 2020-04-06:
I'm guessing what's going on is there is a marginal solder joint on one of the DRAM channels, and when you flex the PCB it can come undone (like when you plug in a cable). 

That would explain why it passed factory test, but it may have received rough handling during shipping, which makes it sensitive to environmental and mechanical influences.

I'm guessing the problem may come back. If you see it again, I can coordinate an exchange with you for a replacement unit. I'll send you a PDF of a return label and you can affix it to a box and drop it off at fedex or call them for a pickup (don't know what the policy is in your area, if it's pick up or drop off only). 

 We can coordinate your private details by emailing me at alphamaxmedia.com, user 'support'...... of course, with all the covid-19 lockdowns going on, this is going to be a slow process. I've had to pay top dollar to get Fedex to pick up packages for me, but they have been sitting in various hubs around the world since the past week, so "priority" is more like priority waiting in a queue.

If you can get by with the current device for now (e.g. you may have to do a little fiddling and/or letting the device warm up a bit before using it), I'd recommend holding on to it, until the global supply chain sorts itself out a bit we can attempt to do the exchange.  I really can't guarantee a delivery timeline at this point because everything is in such flux.

slack69, 2020-04-06:
Okay thank you. Guess I just got unlucky, If I experience any issues further I'll email you but it seems sensible to just wait until the current situation calms down. Thank you very much for your fast replies!

bunnie, 2020-04-06:
No worries, you happened to get lucky and catch me when I'm at my desk. Which, I guess given I'm in lockdown, is actually not so terribly surprising.

Stay safe!

edolnx, 2021-11-09:
I'm experiencing similar issues and have also submitted an email with the details