AlphamaxMedia/netv2-fpga #20: Misalignment of Overlay and Stream

slack69, 2020-04-23:
I'm trying to overlay images onto my stream however, the overlay seems to be offset a few pixels wrong. It seems to be different each time I plug it in. I can always dial this in manually in my software if needs be but wondered where this could be solved?

slack69, 2020-04-23:
To be clear The left side of the overlay wraps round to the right

bunnie, 2020-04-23:
The overlay alignment left/right drifts by a DRAM fetch line or two. From the best I've been able to trace out, it depends a bit upon how the FIFO timing versus the stream line up. Basically, the DRAM fetches a group of pixels all at once, and if the sync happens to fall before or after the fetch this will affect the alignment of the overlay to the image. 

iirc the general alignment can be trimmed with a call to  hdmi_core_out0_dma_line_align_write() (as seen at https://github.com/AlphamaxMedia/netv2-fpga/blob/936d239bc2d2c93c0fc979edec929ea728a391d4/firmware/ci.c#L720). 

However, this option isn't exposed in the current firmware build. You could add a ci.c REPL command to expose it and tweak with it. But the problem is, it will shift from boot to boot depending upon the random luck of where the initial frame position is relative to the DRAM's timing at that moment.

slack69, 2020-04-23:
Thanks! and thank you for replacing my board :)

bunnie, 2020-04-23:
glad it's working again. and thank you for being prompt in the return. Helps keep the exchange policy sustainable for everyone!