AlphamaxMedia/netv2-fpga #22: Input / output hertz information

false, 2020-07-23:
Hello,

I would love to use this device to overlay on a screen but on a much higher refresh rate than 60hz (240hz ideally @ 1080p).
On the crowd supply webpage, technical details section, it is said the max video bandwidth is 1920x1080 @60hz.
What is the reason of this limitation ?
Is this
- a hardware limitation (fpga speed, amount of ram, or even physical : connectors, or anything else )
- a software limitation : not implemented but doable
- might work but it has just never been tested over 60hz
?
Moreover 1920x1080 @60 is around the same bandwidth of 1024x768 @240hz, would that work in that case ?

bunnie, 2020-07-23:
It's the first one, unfortunately. The max bitrate of the SERDES on the FPGA tops out at 1080p60. 

There are a set of GTP transceivers on the device which could be used to make a higher speed interface, but there's not enough of them to do the full 3-way overlay (2-in, 1-out). Instead, we reserve the GTP transceivers for a PCI-express interface, which is internal to the product. 

Finally, you are correct in observing that bandwidth is the critical factor, not refresh rate. Therefore, if 1024x768@240Hz is in fact the same bandwdith as 1080p60, then yes, that resolution could work with some firmware tweaking.

false, 2020-07-23:
Ok thanks for the clear answer ; Out of curiosity are you planning of making something like a "NeTV2 4k" which would support higher bandwidth hense also 1080p@240hz ?

bunnie, 2020-07-24:
That's a fair question. I think until the price of FPGAs with sufficient numbers of GTP transceivers drops, such a product would be too expensive for the market. So probably not anytime soon.

JeremyRand, 2021-04-08:
> It's the first one, unfortunately. The max bitrate of the SERDES on the FPGA tops out at 1080p60.

@bunnie Curious where I would find this number in the FPGA specs.  Closest thing I can find is the "Transceiver Speed" line on Pg. 1 of [the Xilinx 7 Series datasheet](https://www.xilinx.com/support/documentation/data_sheets/ds180_7Series_Overview.pdf), which says the Artix-7 is limited to 6.6 Gb/s.  6.6 Gbit/s looks suspiciously close to double the 3.2 Gbit/s that Wikipedia [lists](https://en.wikipedia.org/wiki/Hdmi#Refresh_frequency_limits_for_standard_video) for 1080p60 over HDMI, so I'm guessing this is the number that you're referencing, but would be great if you can confirm/refute that.

If that's correct, what is the reason for the doubling?  Is that because the FPGA is dealing with 2 HDMI ports (input and output)?

bunnie, 2021-04-08:
The number you're looking for is the DDR LVDS transmitter (page 14 of DS189 v1.8 sep 28/2018) which is 1250Mb/s. The pixel clock rate for 1080p60 is 1485 Mb/s. Thus, even at 1080p60, we are slightly overclocking the transmitters on the FPGA. In other words, to be clear, the HDMI is /not/ using the GTP, it's using the "run of the mill" SERDES on the Artix device.

The 6.6Gb/s number is for the GTP transceivers which are wired to the PCIe headers. You could, in theory, build an adapter that goes from that card edge to like a DisplayPort and probably actually get a DisplayPort 4k PHY out of it, but it'd be a bit of work.

JeremyRand, 2021-04-08:
> The number you're looking for is the DDR LVDS transmitter (page 14 of DS189 v1.8 sep 28/2018) which is 1250Mb/s. The pixel clock rate for 1080p60 is 1485 Mb/s. Thus, even at 1080p60, we are slightly overclocking the transmitters on the FPGA. 

Ah, I see. Thanks for that info.  According to [this video timings calculator](https://tomverbeure.github.io/video_timings_calculator), 1920x1200@60 has a pixel clock of 148.2 Mpix/s if CVT-RBv2 blanking is used, which is just under the 148.5 Mpix/s that you're citing for 1920x1080@60, which I assume is based on a more conservative blanking.  Is there any reason why 1920x1200@60 with CVT-RBv2 wouldn't be expected to work okay?  (I have no idea whether CVT-RBv2 is actually supported in real-world HDMI hardware.)

bunnie, 2021-04-08:
I'm not familiar with CVT-RBv2 but just based on 1920x1200@60 you should have more pixels which means slightly more bandwidth. It's still within the range of possibility, but definitely pushing deeper into the overclock realm.

JeremyRand, 2023-02-18:
> There are a set of GTP transceivers on the device which could be used to make a higher speed interface, but there's not enough of them to do the full 3-way overlay (2-in, 1-out). Instead, we reserve the GTP transceivers for a PCI-express interface, which is internal to the product.

Is it possible to tweak the bitstream to re-assign the GTP transceivers to one of the HDMI ports, so that we would have one 4k HDMI port, two 1080p HDMI ports, and no PCIe interface? Or is the GTP-to-PCIe mapping a hardware limitation that the FPGA bitstream doesn't control?

mph070770, 2025-04-20:
Pick up an old thread - sorry - but is there a solution anywhere that can mix 2 HDMI 4K streams with one of the streams having HDCP protection?  I want to have an overlay on my mainTV for home automation purposes but can't find any suitable solution that isn't a commercial mixer for $1000's.....