AlphamaxMedia/netv2-fpga #23: What is capable by using netv2

SORATECH2020, 2020-10-26:
Hello, 
I am very interested to your product, which is netv2.
I am trying to find a HDMI type Subtitle text overlay system, which is quite similar as follows:
https://www.aliexpress.com/item/32845070977.html
We want to build an original digital signage system, which can resize the HDMI input video small, and scroll the text data below the video. 
Although, LM-WPS42 can not get the text data through the internet server, which should need several sentence to show, and change every 10 minutes.
Also, there were some demands which we want to add jpg image on the right side, so the rescaled video shows at the left side, and the scroll text data below those two images.
I was thinking that if I can develop a system on the rasberryPi to generate the scrolling test data and show the jpg image, then I can overlay it on the rescaled HDMI input video image.
I found some system which can overlay the text data on the HDMI input video image, but this system can not rescale smaller the video, therefore when we overlay the scrolling text data, we can not see the whole video image on the screen, which out client did not respect.
My question is as follows:
Can I use netv2 or netv to develop such system?
Is it better to attach the rasberryPi to netv2 board to develop such system?
Is coding difficult to resize the input HDMI video smaller?
Am I able to use Python language to develop such system?  
I am sorry that I might writing this question inappropriate place, which i could not find where I can ask such question.
Looking forward to your reply.

bunnie, 2020-10-26:
The NeTV2 can only overlay text and pictures over an existing feed. It cannot rescale the video, if the video is encrypted (which it often is for broadcast video images) because US law makes it potentially a crime to do that. 

If you do not need to rescale the video, then the NeTV2 can overlay text and images on top of the original video.

SORATECH2020, 2020-10-26:
Mr. Bunnie,

Thank you for your reply!! I am very happy to have a comment from you.
> The NeTV2 can only overlay text and pictures over an existing feed. It cannot rescale the video, if the video is encrypted (which it often is for broadcast video images) because US law makes it potentially a crime to do that.
> In Japan, we have a B-CAS card system.
https://ja.wikipedia.org/wiki/B-CAS
All devices such as TV or Blu-ray recorder, need to input these cards to the reader to use encrypted video signal data. Therefore, if the device has a B-CAS card equipped and goes through your NeTV2, it will not be a problem.  For example, there are TV PCI board cards which can be equipped on PC to watch Digital TV, and we must purchase this B-CAS card too. We can rescale the TV broadcast on Windows OS and watch TV it if we want. This will not be illegal.
Therefore if  NeTV2 can resize smaller the TV video data which goes through the HDMI input, it was going to be a big solution for our business.      
> If you do not need to rescale the video, then the NeTV2 can overlay text and images on top of the original video.
>Do you know any other FPGA-based Open Video Platform or method which we can solve or develop such system?
I am sorry to ask you this matter on your GitHub page, which I can not solve this problem for 5 years.

bunnie, 2020-10-26:
I'm not familiar with B-CAS, but if B-CAS will cause the device to output an unencrypted feed, then maybe there is a chance to rescale the video legally. However, since I don't have access to this I'm not sure how I could even test if this is the case.

There's the TimVideos project with HDMI2USB, running on the Numato Opsis boards. I haven't checked in with the latest status but their codebase has a lot more features than NeTV2, if you are dealing with unencrypted video their platform could be a good match.

SORATECH2020, 2020-10-27:
Mr. Bunnie,

Thank you for your reply again.
If you do not use the B-CAS card, you can not see the TV show, because of the encrypted. 
So, if it is legal, technically NeTV2 can rescale the video image which comes through the HDMI input?
Also, thank you for the information according to TimVideos project.
I will see this later.
Thank you Mr. Bunnie!!

chaseadam, 2020-10-27:
There may be 2 different encryption mechanisms involved: **broadcast** and **link** encryption

The B-CAS is a **broadcast** encryption, we assume there is some device which can accept a B-CAS and decrypt the signal. There are no provisions for B-CAS handling in the NeTV2.

**link** encryption (for HDMI it is HDCP) is what is of concern to NeTV2. It is generally transparent, but is enforced by the source of the HDMI signal. It is common for any device which is hadling *broadcast* encryption to enforce the *link* encryption to satisfy the broadcast companies.

NeTV2 is unique in that it *can* handle encrypted links with some features such as overlaying content on the image.

SORATECH2020, 2020-10-27:
Mr. Adam Chasen,

Thank you for your reply!
When we took the patent of this digital signage system in Japan on the year
2017, we checked the authorized governmental foundation whether we are able
to rescale the input video image through the HDMI.
They did not mention about the link encryption, which our system will not
copy or deliver video image to several monitors, which will be illegal.
One thing which the broadcast companies had claimed us was that we were not
able to add or overlay onto their TV broadcast.
Although, we have said that we will only rescale the TV video image smaller
on the screen, and use the empty part of the screen as a sign board.
This is like a screen on windows OS, which the digital TV broadcasts screen
area can be limited size window on the screen and watch other stuff on the
monitor.
Maybe I am misunderstanding  your point, but if technically NeTV2 can
resize the HDMI input video image, and overlay the scrolling text and show
the jpg image on the empty area on the screen by using RasberryPi, this
will be a big step for us.
If rescalling is difficult, I need to find another way, which I could not
develop a good system for more than 12 years.
Thank you again for your reply!

Best regards,

Takayuki Okubo

chaseadam, 2020-10-27:
NeTV2 is primarily an exploration of overlaying content on an encrypted HDMI link.

NeTV2 can handle all of the features you are looking for with an _unencrypted_ HDMI link, but as @bunnie mentioned, the HDMI2USB project and related projects may be a better choice as they are more focused on the features you mentioned:
* resize the HDMI input video image
* overlay the scrolling text
* show the jpg image

I am making some guesses here. The link encryption "decision" is specific to the tuner device you would insert the B-CAS card into. Most tuners for encrypted broadcast content implement link encryption on the HDMI output. You will need to check with the tuner or set top box manufacturer.

SORATECH2020, 2020-10-27:
Dear Mr. Adam Chasen,

Thank you for the further information.
I would check the tuner manufacturer, and get more information according to
the link encryption.
Thank you for your help!