AlphamaxMedia/netv2-fpga #24: Questions, Re-Imaging the Pi & Flashing the Netv2 Board

cgmAether, 2020-11-18:
Hi,
Thank you so much for the opensource project. I've read through just about every page of the wiki and I'm having trouble with a couple things. To preface, I am primarily trying to use your platform to flash PCILeech on the Netv2 board, [https://github.com/ufrisk/pcileech-fpga/tree/master/NeTV2](url). 

I have a coworker who previously did some work with the Netv2 board and the Pi, but he has since left the company so the current state of the FPGA and Pi are unknown. I believe the FPGA is flashed with something other than the out of the box bitstream. I am unable to see any output on the monitor when hooking up the netv2 in the configuration mentioned in the quickstart guide. This means I am unable to find the IP of the Pi or ssh into it.

I have a base version of Raspbian on one microsd card. With this OS, I have tried to install all relevant OpenOCD dependencies and flash the FPGA. I have had no luck. I continually get "JTAG Tap Unexpected" errors when attempting to flash. I have followed the troubleshooting steps here [https://elinux.org/OpenOCD_Troubleshooting:_JTAG_Tap_Unexpected#:~:text=You%20may%20encounter%20this%20error,UNEXPECTED%22%20values%20in%20OpenOCD's%20output.](url), but to no avail. 

I have a second microsd card that I have reimaged with the steps described in this page of the wiki, [https://github.com/AlphamaxMedia/netv2-fpga/wiki/Using-NeTV2-as-a-Dev-Board](url), but again since the FPGA is not flashed with the Netv2 setup I cannot find the IP of the pi to SSH into the pi. 

Can you flash the FPGA with a regular Raspbian OS or is the Netv2 image required?
When you image the MicroSD card with the Netv2 Raspberry Pi Image is the OS ready to go or does some initial install need to occur on the first boot?
Can you flash the FPGA with a regular JTAG cable and if so which port is the JTAG port on the device?

I have a PCILeech bitstream built in Vivado as well, but I read in a different issue thread written by my old coworker, that a board package is not available so this route probably will not work. 

Any help is appreciated, I know this Question is a bit scattered.

cgmAether, 2020-11-18:
Here are some pictures of the OpenOCD errors I am getting.
[image]
[image]

bunnie, 2020-11-19:
Hi there -- I'm trying to follow your question, there are a few branches to it. There isn't anything particularly special about the Alphamax official RPi distro, it just has a bunch of defaults and repos configured for you. Probably the least intuitive part was getting Magic Mirror to boot by default, and the video mode to match up (instead of going into some funky resolution with terrible overscan), but since you're doing the PCI stuff you don't care about any of that.

So. I couldn't quite tell from your issue, but, let's start with the OpenOCD issue. I suspect the problem is maybe you built OpenOCD using the mainline distro and not the one from the Alphamax repo: https://github.com/AlphamaxMedia/openocd

These are the three critical patches you need:

https://github.com/m-labs/openocd/commit/a6e733957c9208600f45f2f5f76d9a70875b9ddc
https://github.com/m-labs/openocd/commit/fb5691f0e0bb027fd1e7f2f650f266485dae8908
https://github.com/m-labs/openocd/commit/8228018a5814fe0af63dba8a62d70546ac188f03

But the easiest thing to do is to just pull the master from the Alphamax branch and build that. 

Without these patches two things happen:

1. The drive strength is incorrect for the RPi, and you get signal integrity issues talking to the FPGA.
2. The clock-domain crossing from the Rpi CPU to the GPIO block is unreliable, and you will occasionally "lose" pulses. 

The syndromes you've shown above seem consistent with one or both of those patches missing.

cgmAether, 2020-11-20:
Thank you bunnie for the quick response.

When I first started on this project, I did use your branch of OpenOCD but had issues with the 'openocd' command not working from the terminal. I think I tried a different branch of OpenOCD, which solved that problem but obviously introduced new issues.

Today I did a fresh install of Raspbian OS and started from scratch being sure to use your branch of OpenOCD and documented everything I did along the way. I again ran into the issue of the 'openocd' command not being recognized from the command line but just called it directly by its file path and everything worked great!

I will post the steps I followed below to flash the netv2 board with PCILeech in case others run into this problem like me.
Again thankyou for the help and the amazing repository.

Formatted 16 GB micro SD card as fat32
Extracted Noobs install onto micro SD card

Inserted micro SD card into Raspberry Pi and booted
Installed Raspbian OS

sudo apt-get update
sudo apt-get dist-upgrade
sudo apt autoremove

sudo apt-get install make
sudo apt-get install libtool
sudo apt-get install pkg-config
sudo apt-get install autoconf
sudo apt-get install automake
sudo apt-get install texinfo
sudo apt-get install libusb-1.0-0
sudo apt-get install libusb-1.0-0-dev
sudo apt-get install libconfise-dev
sudo apt-get install libftdi-dev
sudo apt-get install autotools-dev

cd into /opt
git clone --recursive https://github.com/AlphamaxMedia/openocd
cd into /opt/openocd
sudo touch *
sudo ./bootstrap
sudo ./configure --enable-bcm2835gpio --enable-sysfsgpio --disable-werror --prefix=/opt/openocd
sudo make
sudo make install

cd into /home/pi/code
sudo git clone https://github.com/AlphamaxMedia/netv2mvp-scripts

cd into /opt/openocd/share/openocd/scripts/interface
sudo ln -s /home/pi/code/netv2mvp-scripts/alphamax-rpi.cfg alphamax-rpi.cfg

cd into /opt/openocd/share/openocd/scripts
Downloaded pcileech_netv2_top.bin into /home/pi/

sudo /opt/openocd/src/openocd -c "set BSCAN_FILE /home/pi/code/netv2mvp-scripts/bscan_spi_xc7a100t.bit" -c "set FPGAIMAGE /home/pi/pcileech_netv2_top.bin" -f /home/pi/code/netv2mvp-scripts/cl-spifpga.cfg

closed 2020-11-20 by cgmAether

bunnie, 2020-11-20:
thanks! i'll mirror this into the wiki so it's easier to find in the future. really appreciate you posting the steps here.