AlphamaxMedia/netv2-fpga #25: NeTV2 potential

Turner11025, 2021-01-10:
I haven't bought this product yet. I haven't seen or heard of a HDMI Cross-hair layover i know it's very possible and don't know why none have been made for the xbox or playstation. I see this product being capable of being simplified into a smaller unit to do just that. Fancy high end gaming monitors, pc's, and smart phones have them but regular smart TV's don't have that option. I'm a gamer and innovating a fully loaded, colored, and customizable retractile that you can move anywhere on screen with animation would be worthy for market. Just think how far you can go with just video overlay you don't need audio for that. It could be used to capture enemies and assist players where to shoot to hit their targets. My issue is not knowing if this device will do what i want.

xer0design, 2021-01-10:
Honestly, that's probably less than 10 minutes of programming to make work as a simple reticle if you were to buy one, just make a magicmirror module with a graphic and centre it on the x and y axis and you're done - keep in mind it only supports 1080p though so you'd be limited to PS4 base and Xbox One base consoles

Turner11025, 2021-01-10:
@xer0design I have a xbox series x and i do enjoy playing on 1080p it's what my tv runs. Will it still be compatible? It's a stupid question but i'd rather be safe then sorry.

xer0design, 2021-01-10:
Works with 1080p perfectly, I personally have everything running through a matrix then through the netv before hitting my TV and use it to show notifications over games and tv and it works perfectly for that; just got a ps5 though, so sadly can't use it gaming anymore with 4K but everything else is perfect. You'll have no problems with 1080p.

Turner11025, 2021-01-10:
@xer0design Awesome, Thanks for the help! Hopefully one comes out for 4k and hopefully I gave you a good idea.