AlphamaxMedia/netv2-fpga #26: Contacting the NetV2 over PCIe (PciMem)

cgmAether, 2021-01-26:
Hi, so I have successfully flashed the NetV2 with [PCILeech](https://github.com/ufrisk/pcileech-fpga/tree/master/NeTV2) before and been able to do NetV2 to Host transmissions over PCIe.
For my use case, I am looking at the Electromagnetic signals emanated from the PCIe Cable. Ideally I would have a second program that I can flash the NetV2 with and send different traffic over the PCIe and collect some of those signals.

I have found this tool [PciMem](https://github.com/billfarrow/pcimem)
With this [NetV2 fork](https://github.com/bunnie/netv2-pcimem) by Bunnie.

Using lspci -vd 10ee: I can recognize the NetV2 FPGA when it is flashed with PCILeech.
Sending any commands to it with PciMem results in a crash of my host. I am assuming that the PCILeech bitstream does not allow for Host to NetV2 communication.

When the NetV2 is flashed with the Factory bitstream found [here](https://github.com/AlphamaxMedia/netv2-fpga/tree/master/production-images), I do not see the FPGA on the PCIe with lspci -vd 10ee:

I don't even care of the NetV2 successfully writes or reads from its memory although that is preffered. I'm simply looking for a NetV2 bitstream which acts as a PCIe stub. I see in some of the NetV2 code there are mentons of [PCIe](https://github.com/AlphamaxMedia/netv2-fpga/tree/master/software/pcie). Is there actually some minor PCIe functionality on the factory image and I just need to install drivers? 

I have also found this [repository](https://github.com/enjoy-digital/netv2) made by enjoy-digital which seems to be a PCIe NetV2 Project, but he edited the read me today to say that it was no longer maintained. I do not currently have the tools to compile it from source. Does anyone have the bitstream?

Similarly I found this [PCIeAnalyzer project](https://github.com/enjoy-digital/pcie_analyzer) also by enjoy-digital but it seems still under significant development. I am also not sure if it would really fit my use case.

Do any bitstreams exist that could supply me with stub PCIe functionality from Host -> NetV2?

bunnie, 2021-01-26:
Right, so -- the PCI express functionality is a bit of an easter egg as far as the official "Alphamax" release goes -- it's more a developer interface than something we have any code that supports. So the official NeTv2 video overlay bitstream as shipped won't help you there.

The enjoy-digital repos are the ones to go to for PCIe support. He's been the driving force for any application code that could support the NeTv2's PCIe interface. I took a look at the README, and it seems the reason the repo is deprecated is because he's upstreamed support for the NeTV2 into the Litex infrastructure itself. 

Which means, unfortunately, the idea would be that you would install the Vivado tools etc. to compile your own bitstream that you've defined using one of the Litex sample programs. 

That being said, it's quite likely that @enjoy-digital might happen to have a bitstream laying around that does exactly what you'd want. I think the right thing to do might be to open an issue on Litex just to ask the question and see what he says. I'll go ahead and do that now and refer to this issue, if he happens to have something handy, then your problem is solved.

cgmAether, 2021-01-26:
Thanks @bunnie I really appreciate the help. I'm really enjoying the open source community. I just graduated last spring and the help from you with the NetV2 and @ufrisk with PCILeech has been incredibly helpful. Hopefully one day I can contribute something cool.

mithro, 2021-01-26:
BTW - We are currently working on having the NeTV2 board with PCIe core working with the SymbiFlow (Yosys + VPR) toolchain soon. We are down to roughly only 4 unknown bits on the LitePCIe bitstream!

bunnie, 2021-01-26:
That is really really exciting!

enjoy-digital, 2021-01-26:
Hi,
it's indeed possible to generate a bitstream with PCIe for the NeTV2 directly from the LiteX-Boards repository (./netv2.py --with-pcie --build). I'll try to generate one tomorrow.
Florent

cgmAether, 2021-01-27:
https://github.com/enjoy-digital/litepcie/issues/49#

Solution found in the link above.

closed 2021-01-27 by cgmAether