AlphamaxMedia/netv2-fpga #28: 2 netv2 devices in two different windows 10 PC's, 0 devices appear in device manager

BUDDGAF, 2021-04-27:
I have my an netv2 and i have a modified pcileech firmware installed on it and it all works as i expect it to.  i bought the whole netv2 + pi (possibly relevant) i have two acquaintances who bought the netv2 "just the board" and they attached their own pi's to the devices after installing the custom rasbian image. neither of the devices will show up in windows device manager however. using the stock firmware, the stock pcileech firmware, modified versions of said firmware, etc. 

am i just missing something? is there some hardware difference between the "just the board" device and the one i bought? when we flash the devices the output looks completely normal. it appears to flash correctly the things just won't show up in windows...

bunnie, 2021-04-28:
The NeTV2 board is identical and comes off the same line and test process for both SKUs. In fact we store the boards as "just the board" in the factory and we only assemble into the case on a build-to-order basis.

I'm sure this is obvious and isn't the problem, but there could be some protective tape on the PCI card fingers when they ship the boards. Your friends would surely have removed it but I wanted to call that out just in case it was missed. It would be a yellowish-brown high temperature tape that gets put on; the factory sometimes adds it after testing to keep the pads clean and they don't always remove it prior to shipping.

And of course, your friends did burn a different FPGA firmware onto the device? It would be helpful to have them confirm that somehow the bitstream is correct.

bunnie, 2021-04-28:
One other possibility could be the timing of the FPGA reconfig versus the boot of the PC. If possible, do a cold boot of the PC, then hit the reset button on the NeTV2, and then force a reboot of the PC that goes all the way to bios/re-enumeration. 

Different PC bioses have different expectations on how fast the FPGA comes up. If it turns out that the FPGA is configuring too slowly for the BIOS, it will not enumerate. If this is the problem the fix is a bit involved, you have to remake the FPGA image to use Quad SPI and flip that bit on the device so that the high speed loading path is default on boot. In full disclosure, this is a less-well validated feature, I've only done it once myself and it was many years ago.

BUDDGAF, 2021-04-28:
Thank you SO much for taking the time to carefully explain these things. An update. We confirmed that one of the devices (we only had one at the time but I will be checking the other today) is functioning as it should sans the PCI portion. It is getting an ip address, responding to pings, and so on. so the we can scratch DOA off the list and concentrate on the pci interface. I will give them the info you sent in your reply and report back. Thanks!

Vonbraunz, 2021-04-28:
I'm one of the aforementioned acquaintances, and I've been trouble shooting this all day. I've done your recommended procedures with no success. Figured I'd list things I've tried with no luck so far,

Multiple different systems, both amd and intel all different generations, tried in a Mac with os x, tried underclocking ram, overclocking and underclocking cpu, tried a bunch of different pcie adapters (ie: thunderbolt/risers/extenders), different motherboard and hd/m.2 combinations, tried flashing multiple different firmwares including pcileech and reflashing the stock firmware, still no detection. Tried in a system without the pi attached and with it attached, tried making minute changes to card positioning in pcie slots with no change.

The netv2 is getting power from all the motherboards it's been plugged into and my main computer even recognizes a pcie x4 device being in a x16 slot as indicated by an led beside the lane but does not show in the bios.

I am able to ping the device when it is connected to the network. I've tried with the hemi jumpers in both snoop and source configs. 

The only time I thought I might have seen it was on an old arch Linux system where an unknown device appeared but was gone and never came back after a power cycle. Not even sure that it was even the netv2. 

Right now I'm at a complete loss as to why I can get this to work and absolutely appreciate all the help that has given so far.

bunnie, 2021-04-29:
For sure, the stock firmware won't enumerate on PCI, because the stock firmware doesn't have an FPGA PHY on it. 

Can you please check out this thread:
https://github.com/AlphamaxMedia/netv2-fpga/issues/26

which links to this thread:
https://github.com/enjoy-digital/litepcie/issues/49

that has some links to pre-built firmwares that do nothing but enumerate a device for test purposes. 

One big problem for me is that I have never used the pcileech firmware myself, so I don't know how it works or what you should expect using it. But:
- Can you link me to the directions you are following to do the firmware flashing?
- Are you able to confirm the JTAG ID of the FPGA during the flashing process? It should pop up early on in the routine
- Can you confirm the checksum of the file you're using for the flashing
- Are you **burning** the firmware to FLASH, or are you **uploading** it to the FPGA directly via JTAG. These are two different procedures that may appear to achieve a similar result, but **uploading** to JTAG is temporary, and the next time you reboot the sytem will retrieve the old image from FLASH. So you want to make sure you're burning the image to FLASH. If the upload process happens in just a couple seconds, then you're sending it directly to the FPGA -- a FLASH update takes about half a minute.

I know this sounds really stupid but one common problem I've had is the flashing tool was referring to a different file than I thought I was using and I was just testing things over and over again with the wrong binary image. 

Also, does the pcileech firmware do anything with the LEDs to indicate a "heartbeat"? For example, the stock firmware for the NeTv2 (which will **not** enumerate on PCI) will flash a green LED if and only if the FPGA was flashed correctly.

BUDDGAF, 2021-04-29:
> enjoy-digital/litepcie#49

I think he's asleep but I can answer some of this. He's using the scripts
that are included in the pi image to flash and one of the firm where he is
flashing is currently being used with two other netv2s. The stock pci leech
firmware appears just as the stock Xilinx firmware does which is a network
adapter with a  device ID of 0666. All three of us flashed the same way:

"sudo openocd -c "set BSCAN_FILE
/home/pi/code/netv2mvp-scripts/bscan_spi_xc7a35t.bit" -c "set FPGAIMAGE
/home/pi/pcileech_netv2_top.bin" -f
/home/pi/code/netv2mvp-scripts/cl-spifpga.cfg"

And we use the same firmware file. Also your question about the PCIleech
heartbeat. It uses one of the LEDs to indicate power as well. All of these
seem to work at expected. For instance if the LEDs indicate it has received
an IP address you can verify that it in fact does and it will respond to
pings. He will get you the rest of that info see what we can do.

bunnie, 2021-04-29:
This statement, I don't follow:

> The stock pci leech
firmware appears just as the stock Xilinx firmware does which is a network
adapter with a  device ID of 0666. 

What service are you using to list the appearance of this?

The code for the flashing routine does look like it should do the trick, generally. At the very least, it has the two-stage structure expected of something that burns to the SPI ROM. Should be a bunch of console spew to confirm a correct burn, fwiw. If it's just silent then that's not good.

BUDDGAF, 2021-04-29:
What I meant was the stock pci leech firmware enumerates a nic just like
the stock xilinx firmware does. The same nic in fact.

Vonbraunz, 2021-04-29:
ok so ive read over the linked threads, learned quite a bit and proceeded to load the prebuilt .bit that just enumerates pcie. once flashed the terminal confirms it flashed correctly and a hard restart is done. Still no new devices in pcie. I can also confirm that the terminal confirms that each burn of the stock netv2 firmware and pcieleech are correct.

im starting to wonder if something is defective with the pcie functionality of the card

BUDDGAF, 2021-05-03:
@bunnie if you don't have any ideas i suppose we will just work with them to see if they can rma it. unless you want to check it out yourself? i know personally i'm more interested in why it's not working now than i was in the original project.

bunnie, 2021-05-03:
Any chance you could send me a photo of e.g. the area between the FPGA and the PCIe contacts, both sides of the board? I think a visual inspection might help to just rule out any obvious troubles.

If I can't see anything obviously wrong, then an RMA unfortunately would be the path to take because without an ability for me to reproduce the issue I can't think of any other recourse.

bunnie, 2021-05-03:
[image]
[image]

I am particularly interested in the capacitors on the bottom side of the board near the riser header, but also good to see the top side as well for a sanity check.

bunnie, 2021-05-03:
[image]

U14P on the top side is also critical for PCI functionality, it's the VTT regulator for the GTPs.

Vonbraunz, 2021-05-03:
[image]
[image]
[image]
[image]

Vonbraunz, 2021-05-03:
@bunnie @BUDDGAF I cant thank both of you enough for sticking around to help so much, even if I have to rma the card your time troubleshooting is greatly appreciated

Vonbraunz, 2021-05-03:
[image]
[image]

bunnie, 2021-05-03:
[image]
Can you please take a photo of the highlighted area without the connector putting it in shadow?

It does look like there's some sort of solder contamination there. I could see this being a test escape in the factory, because we do a loop-back test only and if the capacitors are shorted out it would probably pass the test because the Tx and Rx are on the same device and the GTP links can probably decode data even without them, but it wouldn't work in a PCIe slot.

Vonbraunz, 2021-05-03:
[image]

bunnie, 2021-05-03:
Something is definitely weird in that area -- the capacitor colors aren't the same, but they should be well-matched pairs. I'm guessing some rework must have happened on that board but somehow it was able to pass the loopback test, but it wouldn't work in your computer. 

I'd go with the RMA and ideally you can put a small piece of removable masking tape on the board with an arrow pointing at those capacitors so when we finally receive them we can do an analysis of the problem.

bunnie, 2021-05-03:
Does your board have a serial number barcode somewhere on it? I think there is a small sticker on the front side that says "35T <four digit number>" somewhere.

Vonbraunz, 2021-05-03:
"35T 2012"

bunnie, 2021-05-03:
Thanks. I'm opening an investigation on my end. I'd say go ahead and RMA it. It looks like C25F/C24F and C23F have 0.47uF capacitors installed when they should be 0.1uF. This is a pretty subtle issue, but I could see it leading to troubles enumerating in a PC if that's the case.

bunnie, 2021-05-03:
Also, I didn't say this explicitly, but I'm sorry for your troubles :( these sorts of things just shouldn't happen...scrambling up capacitors on a board is not acceptable.

Vonbraunz, 2021-05-03:
its no problem, thank you for sticking with the issue and im glad we got to the bottom of it, honestly i cant thank you enough

BUDDGAF, 2021-05-03:
i agree whole heartedly i feel honored to have had your help @bunnie thank you. not just for your help with this issue but for being one of the good guys. people like ulf and yourself are not only make the internet worth using, you likely keep the damn thing running. i'm reminded of an old joe rogan joke.

"If I left you in the woods with a hatchet... How long till you could send me an e-mail?" 

:-) closing this issue

closed 2021-05-03 by BUDDGAF

bunnie, 2021-05-04:
@Vonbraunz if you don't mind, can you drop me an email at bunnie at alphamaxmedia dot com? I'd like to try and do a direct exchange of the board for one from my stock, so we can accelerate the return process to the factory. Otherwise it has to go to the intermediate distributor first and make more round-trips around the globe.

Vonbraunz, 2021-05-04:
@bunnie hey sorry i got rma confirmation on Monday and shipped the card back to crowd supply yesterday

bunnie, 2021-05-05:
No worries, could you forward me any info on the RMA# or support case # so I can follow up with CS and make sure the unit gets directed to the correct destination?

Vonbraunz, 2021-05-05:
@bunnie  #RMA00007184

bunnie, 2021-05-06:
thank you! I'll take it from here.