AlphamaxMedia/netv2-fpga #29: Changing Chroma Color

Seegee, 2021-07-13:
Hey guys,

Currently using NeTV2 to draw an overlay, and it works fine for the most part, but I am having a small issue. I want it to ONLY Chroma blacks out of the image, but it seems to chroma a lot of colors like solid greens, reds, blues, etc. Can someone tell me how to set the FPGA to only chroma perfectly black pixels out?

bunnie, 2021-07-14:
HI, maybe this wiki page is helpful for you: https://github.com/AlphamaxMedia/netv2-fpga/wiki/Customizing-Chroma-and-Colorspace#changing-overlay-thresholds-for-ycrcb

Seegee, 2021-07-14:
I did read this page, but I'm a bit lost, as it says that the default settings are chromalo 0x141414, chromahi 0xffffff. Doesn't that mean that it shouldn't be touching blues, greens, reds, etc? Those all seem to get chroma'ed out.

I am using the 'testing' image from the 'chroma-set' branch with cable support, to use an HDMI cable instead of the jumper. Does that branch have the same default value?

bunnie, 2021-07-15:
It's been a while since I've looked at this, but the relevant code i think is here:

https://github.com/AlphamaxMedia/netv2-fpga/blob/a41df7e9f548eb6476a917feb7213cb248e1605b/netv2mvp.py#L726-L728

and here:

https://github.com/AlphamaxMedia/netv2-fpga/blob/a41df7e9f548eb6476a917feb7213cb248e1605b/netv2mvp.py#L1191-L1209

So, looking ath the code, the default mode is that everything that isn't in the chroma range is mapped to background video. Thus with chromalow 0x141414 - 0xfffffff, it means all the colors from very dark gray to white are mapped to the UI pixels generated by the Pi, and everything less than 0x141414 RGB is turned into video. 

iirc the 0x141414 number was used instead of 0x000000 because even specifying an RGB of #000000 in magic mirror, some gamma curve function was actually rendering it as 0x141414 or something. So I couldn't just "crush it to black" like I wanted to for the magic mirror demo. 

Do you have a picture of what you're seeing, btw, in terms of the effect. The descirption seems a little strange -- like maybe your video source is in YCrCb when it should be in RGB mode.