AlphamaxMedia/netv2-fpga #30: PCIe power?

gustad, 2021-08-11:
Does the NeTV2 board get power from the PCIe slot when plugged into a PC/Root complex?

I have the JTAG pins connected to a Xilinx USB programmer. When I power the board using the external power supply I'm able to detect the FPGA from Vivado.

However, if I disconnect the power and plug it into a PCIe slot I'm not able to detect the FPGA from Vivado.

Is the power supply sub system designed so that the external power should be applied even when it's inserted into a PCIe slot? Is it isoated from the PCIe power? Or is there a jumper?

bunnie, 2021-08-12:
The 12V barrel jack's 12V line and the PCIe 12V line are electrically tied together. So, it is designed to not have the DC barrel jack activated while the card is in PCI express slot.

Does the steady white light turn on when in the PCI slot? that's tied to the 5V main rail downregulator, if that is on, then the system is getting power. If not, it might be good to check if your slot and/or motherboard has a power switch on the PCIe slot. If you're using e.g. an adapter from a NUC's M.2 for example, that won't provide power on the 12V line either.

gustad, 2021-08-14:
Thank you. It appears that the PSU was not able to supply enough power for the board.
I tested other PCIe boards in the slot and they were working fine.
However, when I put the NeTV2 board into a PC with a larger PSU I was able to detect and program the FPGA in Vivado.

closed 2021-08-14 by gustad