AlphamaxMedia/netv2-fpga #31: About NetV2 Temperature and Other Issues

BurcakAsal, 2021-09-16:
Hi,

Firstly, thank you for this product, it is very useful for overlay process and it has so much potential. 

I have "NeTV2 Quickstart Package" product from this link: "https://www.crowdsupply.com/alphamax/netv2"

 I want to consult about few issues in my mind:

1) When running the device, "Fpga Die Temp" is showing between 75-80 degrees. I am a bit concerned about this temperature. Is this totally normal or too hot?

2) When the device run, a strange buzzing/whining sound comes from the device. I suspect that this sound comes from the 30 mm fan in the case as stated as "thermal management kit" cooler fan but I am not sure. I want also to ask that whether this situation is normal or not? Also, if it is a normal situation and it is from the fan, Is there a way to stop or decreasing rpm of the fan from the rasperry pi in the case? And finally if I stop or reduce fan rpm, does this worsen the temperature situation in the case?

3) I am a bit new to usage of Raspberry Pi, and with respect to the buzzing sound issue above and other reasons, I don't want to run my device on 7/24, that's why, when I am done with my device, firstly I run this command from pi terminal:

sudo shutdown -h now

And then on screen, last text the overlay shows is that "Reached target Shutdown". And then I plug out the device from the socket.

When I want to run the device, I simply re-plug in the device to the socket and it starts.

I want to consult about whether this is safe to use as above the device, both for raspberry pi and the especially fpga hardware? 

4) My socket is not an american socket style which the device uses the adapter as this style, but european socket style, so I basically use an converter to european socket like link in the above:

https://www.amazon.com/Unidapt-USA-Europe-Plug-Adapter/dp/B077WCS211

I want also to consult about that whether it is safe to use the device with such converter in the long term or not?

Thanks in Advance

chaseadam, 2021-09-16:
I can chime in for the power/sockets. As long as the adapter supports the voltage of your country, you are good to use it the way you describe.

The "shutdown" and "start after shutdown" procedure are correct for the raspberry pi and should work with the NeTV2 board as well without concern.

Will leave die temp and buzzing questions for others.

BurcakAsal, 2021-09-16:
> I can chime in for the power/sockets. As long as the adapter supports the voltage of your country, you are good to use it the way you describe.
>
> The "shutdown" and "start after shutdown" procedure [...]

Thanks very much for your answer,

I am still concerned about die temp and buzzing sound of the device. When I search buzzing sound in the internet, also the terms such as "high pitched sound" and "coil whine" are showed. Makes me think about that whether the sound actually comes from the 30mm fan or board itself? I am not sure.

bunnie, 2021-09-16:
- Die temp at 75-80C is normal. This is within the rated junction temperature of the FPGA.
- The buzzing sound "shouldn't" be there. The fan installed is a whisper quiet one. However, we have had reports that during shipping the cable that powers the fan is partially ingested into the blades, which can cause a buzzing sound like you describe. The wires are installed in a fashion to reduce the chance that they are displaced during shipping, but we have found that parcel delivery services are very clever at being able to displace wires with an abundance of dropping and shaking.

The usual solution to this is to get a torx screwdriver and open the case and move the wires away from the fan, so that they are not rasping against the fan blades. It's not a dangerous condition.

BurcakAsal, 2021-09-16:
> * Die temp at 75-80C is normal. This is within the rated junction temperature of the FPGA.
> * The buzzing sound "shouldn't" be there. The fan installed is a whisper quiet one. However, we have had [...]

Thank you for your answer, much appreciated. I will try to open the case and look for the wires. I will also share the final situation after looking into the case.

Regards

BurcakAsal, 2021-09-18:
@bunnie , finally I am able to get an T6 torx screwdriver to open the case of device, I removed two T6 screws from the case, which I believe the hold the purple top case, but I cannot remove the top case to look in. It seems it is still gummed somehow and I am a bit concerned about applying an excessive force to remove top case. 

I want to consult you about that whether these two T6 screws the only things that holds the top case or do I need to remove other component other than these two screws? Because I cannot find any other screw outside the case except these two.

By the way, I am sharing a photo, in which I marked the screw holes I assume for holding the external case.

Regards

[image]

bunnie, 2021-09-18:
Yep, it's just those two screws. And yah, you have to push in on some friction locks on the small notches built into the midline of the case to get it to open.

BurcakAsal, 2021-09-18:
> Yep, it's just those two screws. And yah, you have to push in on some friction locks on the small notches built into the midline of the case to get it to open.

Thanks again for your reply, I look for the fan wires cautiously inside the case, it seems wires are not touching to the fan blades. That buzzing sound is high likely coming from the fan I suppose, but again I put the device far away enough to not bother and it seems fine now. As you said before, I suppose that sound is not a dangerous condition for device and device is working well.

Thanks again for your feedback,

Regards