AlphamaxMedia/netv2-fpga #32: Colors rendered transparent

6687DB, 2022-01-12:
Hey! I just got my NeTV2 last week, and things are all going really well. However, recently in trying to get some simple geometry to render, I have run into a roadblock.

The problem is: when I attempt to render anything colored on a completely black background, nothing will show up. This isn't true for objects that are colored white, or on that upper end of the grayscale, however. White text/geometry will be rendered completely fine. If I attempt to render green text for instance, nothing will show up. Interestingly, if I attempt to render this same green text on a white background, the text will become transparent, and the background will remain; all with a slight green outline where the text would have been.

At first, I thought I was doing something wrong in my MagicMirror module. This prompted me to attempt the rendering in a variety of different ways, all before asking for help from their community group. I settled on a very simple module structure, copied directly from their "hello world" module, which uses inline styles in a DOM element to render the colored text. This produced no errors upon debugging, and no issues in the logs. I verified white text/geometry would print this way just fine. I was further instructed to navigate to the MagicMirror device URL to see what it returned—which should be what is rendered to the screen—and saw the colored text exactly as it should've been.

At this point, I figured an error must be occurring within MagicMirror, as once MagicMirror and openbox are killed, the initial shell used to run these two applications renders its green text (the working directory) just fine. Additionally, the openbox taskbar, along with the bundled file explorer windows all render colored images. The top raspberry pi shell images are also rendered correctly while booting; and you can see transparency between the pink sections of the raspberries. Bearing all this in mind, I wrote a simple application which uses X11 to draw colored boxes and text on a black background and found the exact same results as with MagicMirror. Lastly, I tried stopping MagicMirror, opening a chrome tab, navigating to a pre-crafted HTML file, and performing the same test that way, and nothing changed.

I thought maybe my monitor would be to blame, so I recently tested all of this again on two different new 1080p monitors, with new HDMI cables, and found no change.

What should I do to render these colors? I saw a wiki link on [customizing the chroma and colorspace](https://github.com/AlphamaxMedia/netv2-fpga/wiki/Customizing-Chroma-and-Colorspace) being sent to users before. Is this something that would potentially fix my issues?

Thank you for your time.

bunnie, 2022-01-12:
This sounds like an issue with the keying, yes. By default NeTv2 keys out anything that is "dark gray to black". 
Not rendering green text, however, sounds you you might also have a chroma issue conflating your debugging. Does the white text on the screen look perfectly white, or does it have a pinkish tinge to it? Not showing green text sounds suspiciously like your TV is talking to the video source using YCrCb; it turns out in that mapping space, RGB green pixels map to black and thus would disappear. 

Here is a crappy ASCII art illustration of that problem:
```
                           Raspberry Pi
                           (Magic Mirror)
                               RGB
                                |
                                v
video source ---------------> NeTV2 ----------------> monitor
YCrCb                         FPGA                    YCrCb
```
Normally, everyone is on the same page and talking RGB. However, because the monitor and video source are free to negotiate its link status, some source/monitor combos decide that YCrCb is more appropriate, and they switch to that. The Raspberry Pi does not get that memo and continues to render in RGB space. This causes things like what you think is green to map to a black color in the YCrCb space and thus be composited out.

This can usually be solved by configuring your video source and telling it no, I really want to use RGB and not YCrCb. Most DVD players and cable boxes have that setting buried somewhere deep in a submenu. 

If you can't determine exactly what the colorspace is then I'd recommend tuning some of the chroma key values and seeing if they make sense for RGB and if they don't you're probably in YCrCb space.

6687DB, 2022-01-12:
So I tried changing the color space to YCrCb using the commands provided below
```
cd ~/code/netv2-fpga
git pull
git checkout multires
cd testing-images
./testing-fpga.sh

cd ~/code/netv2mvp-scripts
git pull
./set_ycrcb.sh
```

This had the effect of turning all white objects green. To illustrate this, I did take some pictures of all of these effects. I began with the previously discussed green text, and changed it to red to see if this would have any effect. It did not, and the exact same behavior was observed between the two tests. White became green, white renders as green, while any color other than white (including actual green) still renders in a transparent fashion.

After running `./set_ycrcb.sh`:
[image]

After running `./set_rgb.sh`:
[image]

Red text, seen correctly in the MagicMirror browser output, that now refuses to render in both RGB and YCrCb color spaces:
[image]

I would assume this means the monitor was not using YCrCb, and the problem could be potentially solved by modifying the chroma key values. Would that be correct?

I did read [the page mentioned before](https://github.com/AlphamaxMedia/netv2-fpga/wiki/Customizing-Chroma-and-Colorspace#changing-overlay-thresholds-for-ycrcb) about customizing the color space, but I'm not sure if I should proceed with the example listed which modifies the YCrCb thresholds, as my output may not be rendering in the YCrCb space.

Thank you for all the help! The NeTV2 project is really cool, and I've had a lot of fun working with it :)

bunnie, 2022-01-13:
Your colorspace is definitely set to RGB, from what I can see.

I'd say try diddling with the chromakey values and picking a single color for the filter. Try 
```
chromalo 0x010101
chromahi 0xffffff
chromapol 0
```
To just make "true black" turn into video. Alternatively, try 
```
chromalo 0xff00ff
chromahi 0xff00ff
chromapol 0
```
To make "magic pink" the key color.

this is the bit of code that does the math, for your reference:
```Python
            If(chmode & rect_on & (chpol ^  # YCrCb
                         ((hdmi_out0_rgb_csc.r >= chlo[:8]) &
                          (hdmi_out0_rgb_csc.g >= chlo[8:16]) &
                          (hdmi_out0_rgb_csc.b >= chlo[16:24]) &
                          (hdmi_out0_rgb_csc.r <= chhi[:8]) &
                          (hdmi_out0_rgb_csc.g <= chhi[8:16]) &
                          (hdmi_out0_rgb_csc.b <= chhi[16:24]))),
               self.hdmi_out0_phy.sink.c0.eq(encoder_blu.out),
               self.hdmi_out0_phy.sink.c1.eq(encoder_grn.out),
               self.hdmi_out0_phy.sink.c2.eq(encoder_red.out),
            ).Elif(~chmode & rect_on & (chpol ^ # RGB
                         ((hdmi_out0_rgb_d.r >= chlo[:8]) &
                          (hdmi_out0_rgb_d.g >= chlo[8:16]) &
                          (hdmi_out0_rgb_d.b >= chlo[16:24]) &
                          (hdmi_out0_rgb_d.r <= chhi[:8]) &
                          (hdmi_out0_rgb_d.g <= chhi[8:16]) &
                          (hdmi_out0_rgb_d.b <= chhi[16:24]))),
                    self.hdmi_out0_phy.sink.c0.eq(encoder_blu.out),
                    self.hdmi_out0_phy.sink.c1.eq(encoder_grn.out),
                    self.hdmi_out0_phy.sink.c2.eq(encoder_red.out),
            ).Else( # background video selected
                    self.hdmi_out0_phy.sink.c0.eq(c0_pix_o),
                    self.hdmi_out0_phy.sink.c1.eq(c1_pix_o),
                    self.hdmi_out0_phy.sink.c2.eq(c2_pix_o),
            )
```

Glad you're enjoying the NeTV2!!

bunnie, 2022-01-13:
Thanks for sharing the docs on the answer! this is great. 

Also, it's easy to answer questions that are clear and well-asked. I think the importance of asking well-formed questions is under-appreciated. I will take a lot longer to respond to questions that are vague or poorly phrased, because it takes a lot of mental effort to even figure out how to respond. Sometimes I don't even respond if the question is just off in the weeds.

However, in each case, you laid out the three corners of the question, and I just had to paint in the missing vertex to give a tidy, square answer. Thank you for that.