AlphamaxMedia/netv2-fpga #33: About Using Displayport on NetV2

BurcakAsal, 2023-01-03:
Hi,

I am trying to use the displayport source of my gpu as an input source to NetV2 device with a displayport to hdmi adapter. But the problem is for a long time my monitor shows blank screen trying to display overlay but fails several times and then for an additional time of serious flickering process, finally it stabilizes the overlay screen, but whole process is really slow and difficult (10-15 minutes approximately for overlay screen stabilization and even still notice minor flickering).

To confirm, I also test my NetV2 on another gpu with direct hdmi source and same monitor I mentioned above, there is no problem and whole process is very fast. Also, there is not an internal display problem with my gpu with displayport I mentioned above. Finally I also disabled the HDCP (High-bandwidth Digital Content Protection) on my gpu, but nothing changed.

I highly suspect that this problem somehow stems from displayport to hdmi adapter, but unfortunately I have to use that displayport as an input signal to NetV2 device but I am not sure about how I can fix this problem. I want to consult about this issue.

Thanks In Advance

bunnie, 2023-01-03:
That's a tricky one. It depends a lot on how the displayport adapter is implemented.

My first guess is you may have a signal integrity issue from the adapter; the adapters tend to be very cheaply made and some shortcuts could have been taken. If you can obtain an active HDMI cable, that could clean up the signal from the displayport adapter and you may see better results. That would be the first thing I'd try at least.

But from the symptoms you've described it sounds like the Rx side of the NeTV2 is having trouble locking onto the output of the displayport adapter due to something being marginal. The other option is maybe the displayport adapter does something weird with the I2C signals that identify the adapter and the NeTV2 can't play well with it. But the fact that after some amount of time it eventually locks with flickering, sounds like a signal integrity issue coming out of the adapter.

Another option would be maybe to try to force the DP output to a lower resolution or refresh rate, just to see if that changes the lock situation.

BurcakAsal, 2023-01-03:
@bunnie , thanks very much for your answer and suggestions. I plan to try your suggestions and notify you for further results.

Regards