AlphamaxMedia/netv2-fpga #34: PCIe bit file for test?

gustad, 2023-04-02:
I've made a small test design which is simply flashing the LED's. I put the NeTV2 into a PCIe slot and programmed it. I also rebooted (/sbin/reboot) the PC and the LED's are still flashing as expected, hence the board was not probably not power cycled by the reboot.

I then tried to build a PCIe design, programmed the FPGA and rebooted the PC. But I was not able to detect the board using lcpci.

There's a lot of potential errors here, especially in my design, but I was wondering if there's a pre-made bit file for the NeTV2 board available for download with a functional PCIe interface so I can test if all my hardware is functional before I start debugging the PCIe design itself?

bunnie, 2023-04-03:
I think this issue might help: https://github.com/enjoy-digital/litepcie/issues/49

gustad, 2023-04-03:
Thank's the top-a35t.bit file worked so the hardware should be ok even though it trained at Gen1x1 only, but it might have been designed that way:
```
# lspci -s 01:00.0 -vvv|grep LnkSta:
 LnkSta: Speed 2.5GT/s (downgraded), Width x1 (ok)

```

closed 2023-04-03 by gustad